Entware/Entware #914: [Package Request] Jool

zod076, 2023-03-16:
Please consider adding Jool package for AsusWRT-Merlin. There is Tayga package but it is not maintained anymore.

- Jool does NAT64/SIIT and is currently active and maintained.
- Website -- www.jool.mx
- Platform -- AsusWRT-Merlin v388.1

Thank you.

ryzhovau, 2023-05-23:
We've got another NAT64 implementation here. What's the killer feature of Jool?

ryzhovau, 2023-06-06:
No response.

zod076, 2023-09-26:
> We've got another NAT64 implementation here. What's the killer feature of Jool?

Sorry for the late reply. What is the other NAT64 implementation you are referring to ? I only know Jool and Tayga for Linux.

zod076, 2023-11-01:
> We've got another NAT64 implementation here. What's the killer feature of Jool?

Basically better performance over Tayga.

zod076, 2024-01-31:
Hi, been awhile but any news on packaging this for AsusWRT-Merlin ?

zod076, 2024-02-12:
Bump to keep this open. Any updates on this ? Latest jool is v4.1.11.0

zod076, 2024-11-15:
can this somehow be added so I can use this with my Merlin router ?

ryzhovau, 2026-11-04:
[Requires](https://www.jool.mx/en/faq.html#why-is-jool-not-doing-anything) `jool` kernel module.
As long as Entware stays firmware-independent, no any kernel specific parts can be applied here, sorry.

closed 2026-11-04 by ryzhovau