Entware/Entware #917: frp a revers proxy server

sconan32, 2023-03-18:
**Describe the solution you'd like**

https://github.com/fatedier/frp
A fast reverse proxy to help you expose a local server behind a NAT or firewall to the internet.

my router is  asus ax86u. Thank you.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

zyxmon, 2023-03-18:
http://zyxnerd.zyxmon.org/files/frp-go_0.48.0-1_aarch64-3.10.ipk
test and report back plz

sconan32, 2023-03-19:
Thank you. 
it works fine.

zyxmon, 2023-03-19:
https://github.com/Entware/entware-go/commit/fb5d2ad4796f1f7e9c8585254517b87cedc8aad6

closed 2023-03-19 by zyxmon