Entware/Entware #918: mc show error when i try run a command

EugenSoloviov, 2023-03-19:
When i try run a command in mc, mc shows error message:
"The shell is alredy running a command"

[package name]: mc (Midnight Commander)

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - Device [e.g. Zyxel Keenetic Ultra II router]
 - Firmware version [3.4.3.9L-101_289c95a  (Padavan & N56U project community & Linaro & Alxdm) ]

EugenSoloviov, 2023-03-19:
This bug may solve by mc version downgrade.
For solving on my Zyxel Keenetic Ultra II, was executed following commands:

`
cd /media/Main

wget http://bin.entware.net/mipselsf-k3.4/archive/mc_4.8.26-1b_mipsel-3.4.ipk

opkg install /media/Main/mc_4.8.26-1b_mipsel-3.4.ipk --force-downgrade

ln -s /media/Main/opt/lib/libmbedcrypto.so.7 /media/Main/opt/lib/libmbedcrypto.so.3
`

ryzhovau, 2023-03-20:
https://github.com/Entware/Entware/wiki#%EF%B8%8F-package-upgrade-notes

ryzhovau, 2023-05-23:
Can't reproduce it on x64 or aarch64. Looks like device-specific bug.

closed 2023-05-23 by ryzhovau