Entware/Entware #920: bird: Cannot disable transparent huge pages: prctl(PR_SET_THP_DISABLE) failed: Invalid argument

wellloaded, 2023-03-22:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 
I have just installed bird and birdc on my FreshTomato router. Used the sample config to generate something basic in ospf. However running bird -p (to check the config syntax) I get the following error:

```
root@sparrow:/tmp/home/root# bird -p
bird: Cannot disable transparent huge pages: prctl(PR_SET_THP_DISABLE) failed: Invalid argument
```

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 ```
root@sparrow:/opt/etc# cat entware_release
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=8.4.0
gcc_flags=-O2 -pipe -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.23
```

 - Device [e.g. Asus RT-AX88U router]
```
Asus RT_AC66U_B1 
```
- Firmware version [e.g. asuswrt-merlin]
```
Freshtomato 2023.2
```

wellloaded, 2023-03-22:
P.S. just to be precise the software works! However the message in the tile appears every time bird is called from the command line.

zyxmon, 2023-03-23:
It looks like memory configuration problem on your kernel. It is not related to bird.

closed 2023-03-23 by zyxmon