Entware/Entware #922: request for mt-st

AnthraX1, 2023-03-24:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

https://github.com/iustin/mt-st

mt and stinit, used for dealing with Linux-specific tape-drive handling.

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2023-05-23:
I think the number of users of embedded devices with SCSI tape drives is near to zero.
Won't add it for now, sorry.

May be I missing some cool use case for it. Let me know if so.

closed 2023-05-23 by ryzhovau

AnthraX1, 2023-10-13:
> I think the number of users of embedded devices with SCSI tape drives is near to zero. Won't add it for now, sorry.
> 
> May be I missing some cool use case for it. Let me know if so.

entware is not only used on embedded devices. Many people like me install it on NAS such as QNAP devices which have PCIE/ SCSI ports

ryzhovau, 2023-10-13:
It's out of project goal, sorry.