Entware/Entware #938: python 3.10: ModuleNotFoundError: No module named 'email'

blshkv, 2023-05-09:
rdiff-backup used to work just fine. I have updated few packages recently and started to face the following error:
```
/opt/bin/rdiff-backup --server
Traceback (most recent call last):
  File "/opt/bin/rdiff-backup", line 10, in <module>
    from importlib.metadata import distribution
  File "/opt/lib/python3.10/importlib/metadata/__init__.py", line 6, in <module>
ModuleNotFoundError: No module named 'email'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/bin/rdiff-backup", line 13, in <module>
    from importlib_metadata import distribution
ModuleNotFoundError: No module named 'importlib_metadata'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/bin/rdiff-backup", line 15, in <module>
    from pkg_resources import load_entry_point
ModuleNotFoundError: No module named 'pkg_resources'
```

[package name]: python3-rdiff-backup - 2.2.4-1 

**Environment (please complete the following information):**
 - Entware feed: http://bin.entware.net/x64-k3.2/Packages.gz
 - Device Synology NAS synology_v1000_1621+
 - Firmware version the latest version 6

zyxmon, 2023-05-09:
Try this - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

blshkv, 2023-05-09:
So which module do you want to install? It looks like a python install is broken?

From dependencies, I can see the following list:
```
Package: python3-rdiff-backup
Version: 2.2.4-1
Depends: libc, libssp, librt, libpthread, python3-light, librsync
```

or this:
```
RDEPEND="dev-python/pylibacl[${PYTHON_USEDEP}]
        dev-python/pyxattr[${PYTHON_USEDEP}]
        dev-python/pyyaml[${PYTHON_USEDEP}]
        >=net-libs/librsync-1.0:0="
```

blshkv, 2023-05-10:
I think `email` module is a build-in module, shouldn't be included by default?
https://docs.python.org/3/library/email.html#module-email

zyxmon, 2023-05-10:
Correct!
```
 # opkg list python3-em*
python3-email - 3.10.7-3 - Python 3.10 email module
```
Just install python3-email. We will add it to dependencies, if it is missing.

blshkv, 2023-05-10:
ok, I also needed to install '''python3-urllib'''
Now, the error is:
```
 bash$ /opt/bin/rdiff-backup --server
Traceback (most recent call last):
  File "/opt/bin/rdiff-backup", line 33, in <module>
    sys.exit(load_entry_point('rdiff-backup==2.2.4', 'console_scripts', 'rdiff-backup')())
  File "/opt/bin/rdiff-backup", line 25, in importlib_load_entry_point
    return next(matches).load()
StopIteration
```

Any ideas?

blshkv, 2023-05-13:
Hey guys, is any progress here, please? This is kind of important for me.

zyxmon, 2023-05-13:
Try to ask here - https://github.com/rdiff-backup/rdiff-backup or update rdiff-backup tp 2.2.5 using pip.

blshkv, 2023-05-13:
if you have a look at the code ```/opt/bin/rdiff-backup``` you will find that it is not original code, so there is no point contacting upstream.
The code:
```
try:
    from importlib.metadata import distribution
except ImportError:
    try:
        from importlib_metadata import distribution
    except ImportError:
        from pkg_resources import load_entry_point
```

any idea where ``` importlib.metadata``` or ```pkg_resources``` should come from?

zyxmon, 2023-05-13:
This is a standard startup script (openwrt). If it fails - try to make your own script.

zyxmon, 2023-05-13:
```
# opkg install python3-pkg-resources
# opkg install python3-yaml

# rdiff-backup
usage: rdiff-backup [-h] [--api-version API_VERSION].....
```

The-BB, 2023-05-13:
все вопросы к owrt'шникам ([Issues](https://github.com/openwrt/packages/issues)/[PR](https://github.com/openwrt/packages/pulls))... "проблема" в пакете "python3-light":
`pathlib.py` - `from urllib.parse import quote_from_bytes as urlquote_from_bytes`
`metadata/__init__.py` - `import email`
`metadata/_adapters.py` - `import email.message`
минимальный набор для запуска:
`opkg install python3-urllib python3-yaml`

blshkv, 2023-05-13:
Неа, все это пакеты стоят, все равно не запускается.

zyxmon, 2023-05-13:
Почему же у меня запускается? ЧЯДНТ?

The-BB, 2023-05-13:
Изначально вопрос был: "Где модуль потерялся?" ;))
Напишите, как воспроизвести, у меня тоже не вышло уронить...
```
~ # 
~ # ls -al /opt/tmp/backups/
drwxr-xr-x    2 root     root          4096 May 13 14:16 .
drwxr-xr-x    5 root     root          4096 May 13 14:15 ..
~ # 
~ # /opt/bin/rdiff-backup backup /opt/var/run /opt/tmp/backups/
NOTE: Starting mirror from source path /opt/var/run to destination path /opt/tmp/backups
~ # 
~ # ls -al /opt/tmp/backups/
drwxr-xr-x    8 root     root          4096 May  9 11:36 .
drwxr-xr-x    5 root     root          4096 May 13 14:16 ..
drwxr-xr-x    2 nobody   nogroup       4096 Apr 11 08:10 avahi-daemon
-rw-r--r--    1 root     root             6 May  7 09:49 cron.pid
drwxr-xr-x    3 root     nogroup       4096 Apr 11 08:10 cups
drwxr-xr-x    2 root     root          4096 Apr 11 08:10 dbus
-rw-r--r--    1 root     root             4 Apr 11 08:10 dropbear.pid
drwxr-xr-x    3 root     root          4096 Apr 19 15:05 ipp-usb
drwxr-xr-x    2 root     root          4096 May  2 17:21 mysqld
-rw-r--r--    1 root     root             6 May  9 11:36 nginx.pid
srwxrwxrwx    1 root     root             0 May  9 10:45 php-fcgi.sock
drwx------    3 root     root          4096 May 13 14:16 rdiff-backup-data
~ #
```

blshkv, 2023-05-13:
ok, I have removed all python packages, cleaned up all empty dirrectories (it where a lot of them)
installed
```
# opkg install python3-pkg-resources
# opkg install python3-yaml
```
and it started to work.

Thanks!

closed 2023-05-13 by blshkv