Entware/Entware #940: borgbackup 1.2.3-1 ModuleNotFoundError: No module named 'email'

drekbour, 2023-05-17:
**Describe the bug**
Vanilla QNAP install of entware then borgbackup fails

**Environment:**
 - Entware feed: x64
 - Device QNAP HS-251

```
Package: borgbackup
Version: 1.2.3-1
Depends: libc, libssp, librt, libpthread, python3-light, python3-msgpack, libacl, libopenssl, liblz4, libzstd, libxxhash
Status: install user installed
Section: lang
Architecture: x64-3.2
Size: 966812
Filename: borgbackup_1.2.3-1_x64-3.2.ipk
```

```
# borg
Traceback (most recent call last):
  File "/opt/bin/borg", line 10, in <module>
    from importlib.metadata import distribution
  File "/opt/lib/python3.10/importlib/metadata/__init__.py", line 6, in <module>
ModuleNotFoundError: No module named 'email'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/bin/borg", line 13, in <module>
    from importlib_metadata import distribution
ModuleNotFoundError: No module named 'importlib_metadata'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/bin/borg", line 15, in <module>
    from pkg_resources import load_entry_point
ModuleNotFoundError: No module named 'pkg_resources'
```

zyxmon, 2023-05-18:
This problem is similar to https://github.com/Entware/Entware/issues/938
You are to install some missing python3 packages manually. Start with ` opkg install python3-pkg-resources`

ryzhovau, 2023-06-05:
Fixed with https://github.com/Entware/entware-packages/commit/6b7699fe3a7376652e5e296e99e910833b2f6254

closed 2023-06-05 by ryzhovau