Entware/Entware #943: Issue with syslog-ng (maybe others) launching on boot in specific situation

crkinard, 2023-05-27:
syslog-ng and maybe other daemons failing to start on boot when ipv6 is enabled.

 - Entware feed: Not sure how to find this.
 - Device: Asus GT-AXE11000 router
 - Firmware version: asuswrt-merlin 388.2_2
 
Recreate:
-- Factory default.
-- Only options configured: Wi-Fi name and password. 'root' account configured and password. SSHD enabled.
IPv6 enabled with default (native) options. AMTM run and Entware installed. The Scribe installed (which uses syslog-ng).
-- Reboot.
-- 'scribe status' reports syslog-ng is not running.
-- Disable IPv6 and reboot.
-- 'scribe status' reports everything is working fine.

Guess:
Race contrition.
IPv6 startup slowing down boot and messing with Entware services starting properly.
(I had issues with SMB starting on boot some time ago.)

crkinard, 2023-05-27:
No where near an expert at this but I started poking things. Looking for loops or timers that might be related as I think this is a race issue.

Found this:
`/opt/etc/init.d/rc.func

    $PRECMD > /dev/null 2>&1
    $PREARGS $PROC $ARGS > /dev/null 2>&1 &
    #echo $PREARGS $PROC $ARGS
    COUNTER=0
    LIMIT=60
    while [ -z "`pidof $PROC`" -a "$COUNTER" -le "$LIMIT" ]; do
        sleep 1;
        COUNTER=`expr $COUNTER + 1`
    done
    $POSTCMD > /dev/null 2>&1`

That LIMIT=60 variable is LIMIT=10 by default. Increased it to 60 and so far all my reboots with IPv6 enabled have the syslog-ng daemon startting properly.

No idea what's going in in that code there but it seems to have, maybe fixed the issue.

crkinard, 2023-05-27:
After a bunch more testing and file digging this may actually be an issue with a PRECMD script for Scribe that is causing the issue.

Have to break off testing right now. Other people awake in the house that are not to keen on the internet going out when I restart the router.

cynicastic, 2023-05-28:
Not an entware problem, scribe is trying to kill off the ASUSWRT syslogd & klogd, since they don't play nice with syslog-ng, and they are restarting for some reason.

closed 2023-05-29 by ryzhovau