Entware/Entware #945: `deluge` & `deluge-ui-web` don't start since recent IPK update

OneCDOnly, 2023-06-04:
An example:
```
  File "/opt/bin/deluge-web", line 8, in <module>
    sys.exit(start())
             ^^^^^^^
  File "/opt/lib/python3.11/site-packages/deluge/ui/web/__init__.py", line 6, in start
    web.start()
  File "/opt/lib/python3.11/site-packages/deluge/ui/web/web.py", line 69, in start
    from deluge.ui.web import server
  File "/opt/lib/python3.11/site-packages/deluge/ui/web/server.py", line 17, in <module>
    from twisted.internet import defer, reactor
  File "/opt/lib/python3.11/site-packages/twisted/internet/reactor.py", line 38, in <module>
  File "/opt/lib/python3.11/site-packages/twisted/internet/default.py", line 55, in <module>
  File "/opt/lib/python3.11/site-packages/twisted/internet/default.py", line 43, in _getInstallFunction
  File "/opt/lib/python3.11/site-packages/twisted/internet/epollreactor.py", line 19, in <module>
  File "/opt/lib/python3.11/site-packages/twisted/internet/posixbase.py", line 16, in <module>
  File "/opt/lib/python3.11/site-packages/twisted/internet/tcp.py", line 38, in <module>
  File "/opt/lib/python3.11/site-packages/twisted/internet/_newtls.py", line 18, in <module>
  File "/opt/lib/python3.11/site-packages/twisted/protocols/tls.py", line 42, in <module>
  File "/opt/lib/python3.11/site-packages/OpenSSL/__init__.py", line 8, in <module>
  File "/opt/lib/python3.11/site-packages/OpenSSL/crypto.py", line 17, in <module>
  File "/opt/lib/python3.11/site-packages/OpenSSL/_util.py", line 6, in <module>
  File "/opt/lib/python3.11/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 205, in <module>
  File "/opt/lib/python3.11/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 179, in init_static_locks
  File "/opt/lib/python3.11/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 167, in _ensure_ffi_initialized
  File "/opt/lib/python3.11/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 77, in _openssl_assert
cryptography.exceptions.InternalError: Unknown OpenSSL error. This error is commonly encountered when another library is not cleaning up the OpenSSL error stack. If you are using cryptography with another library that uses OpenSSL try disabling it before reporting a bug. Otherwise please file an issue at https://github.com/pyca/cryptography/issues with information on how to reproduce this. ([_OpenSSLErrorWithText(code=310378599, lib=37, reason=103, reason_text=b'error:12800067:DSO support routines::could not load the shared library'), _OpenSSLErrorWithText(code=310378599, lib=37, reason=103, reason_text=b'error:12800067:DSO support routines::could not load the shared library'), _OpenSSLErrorWithText(code=126353445, lib=15, reason=524325, reason_text=b'error:07880025:common libcrypto routines::reason(524325)')])
```

Appears to be due to the old `python3-cryptography` package (3.4.8-3). 

If I manually update `cryptography` to 41.0.1  (and `bcrypt` to 3.1.7) with `pip`, both Deluge packages start fine. However, I'm unable to do this on ARMv5 CPUs without compilation errors - though, this is a separate issue.  

Any thoughts please?

Thank you. :)

zyxmon, 2023-06-04:
Confirmed.

zyxmon, 2023-06-04:
` opkg install libopenssl-legacy` fixes it.  This dependency for python-cryptography is missing

zyxmon, 2023-06-04:
fixed via https://github.com/Entware/entware-packages/commit/a5b9cac834009c57254a7dcd801c867c4fdfd3fa

closed 2023-06-04 by zyxmon

OneCDOnly, 2023-06-04:
Thanks @zyxmon :)