Entware/Entware #949: gcc postrm syntax error

OneCDOnly, 2023-06-10:
Hi guys, just tried an upgrade via sherpa and saw this:
```
= command: '/opt/bin/opkg upgrade --force-overwrite gcc --cache /share/CACHEDEV2_DATA/.qpkg/sherpa/cache/IPKs --tmp-dir /share/CACHEDEV2_DATA/.qpkg/sherpa/cache/IPKs.downloads'
 ! result_code: [255] ***** stdout/stderr begins below *****
//opt/lib/opkg/info/gcc.postrm: line 3: syntax error near unexpected token `||'
//opt/lib/opkg/info/gcc.postrm: line 3: ` || true'
Collected errors:
 * opkg_conf_set_option: Duplicate option tmp_dir, using first seen value "/share/CACHEDEV2_DATA/.qpkg/sherpa/cache/IPKs.downloads".
 * pkg_run_script: package "gcc" postrm script returned status 2.
 * postrm_upgrade_old_pkg: postrm script for package "gcc" failed
Upgrading gcc on root from 8.4.0-5a to 8.4.0-5b...
Downloading http://bin.entware.net/aarch64-k3.10/gcc_8.4.0-5b_aarch64-3.10.ipk
 ! ***** stdout/stderr is complete *****
```

ryzhovau, 2023-06-10:
Thanks again! You can ignore it for now. We'll fix that soon.

OneCDOnly, 2023-06-19:
Also, this error prevents `gcc` being upgraded. 

To upgrade, I had to first uninstall `gcc`, then install it again.

closed 2025-01-17 by The-BB