Entware/Entware #950: lighttpd 1.4.70-1 missing libraries

huibw, 2023-06-12:
lighttpd 1.4.70-1 missing libraries

 - Entware feed: armv7sf-k2.6
 - ASUS RT-AC68U
 - asuswrt-merlin 386.11

A few lighttpd-mod-[addons] for lighttpd 1.4.70-1 are missing the .so files. I upgraded from 1.4.69-1 and lighttpd didn't want to start. Upon further investigation, lighttpd-mod-fastcgi_1.4.70-1_armv7-2.6.ipk was missing the actual mod_fastcgi.so file. The same thing for a few others. When looking at the older versions (http://bin.entware.net/armv7sf-k2.6/archive/) you can clearly see from the file size that a bunch are a lot smaller than their previous version.

The-BB, 2023-06-12:
https://github.com/openwrt/packages/blob/master/net/lighttpd/Makefile#L318-L330

huibw, 2023-06-12:
Interesting. It cannot be a coincidence that the moment they're included in the base fastcgi stops working. Unfortunately I'm not equipped to figure out what else the problem may be with 1.4.70. All I know at this point that it doesn't work, and the error seemed to point in the direction of missing libraries and that's what it looked like what I found. I'll stay on 1.4.69 for now.

- 2023-06-10 18:36:03: (../src/server.c.1909) server started (lighttpd/1.4.70)
- 2023-06-10 18:36:03: (../src/gw_backend.c.526) connect() /tmp/php.socket-0: Connection refused
- 2023-06-10 18:36:03: (../src/gw_backend.c.369) child exited: 127 unix:/tmp/php.socket-0
- 2023-06-10 18:36:03: (../src/gw_backend.c.669) gw-backend failed to start: /opt/bin/php-fcgi
- 2023-06-10 18:36:03: (../src/gw_backend.c.671) If you're trying to run your app as a FastCGI backend, make sure you're using the FastCGI-enabled version.  If this is PHP on Gentoo, add 'fastcgi' to the USE flags.  If this is PHP, try removing the bytecode caches for now and try again.
- 2023-06-10 18:36:03: (../src/gw_backend.c.1741) [ERROR]: spawning gw failed.
- 2023-06-10 18:36:03: (../src/server.c.1913) Configuration of plugins failed. Going down.

closed 2023-06-13 by huibw