Entware/Entware #952: libdht is outdated - please update.

abir1909, 2023-06-21:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - Device [e.g. Asus RT-AX88U router]
 - Firmware version [e.g. asuswrt-merlin]

atkinsom, 2023-06-26:
As per sad-spirit from SNBForums

"Running transmission via startup script from /opt/etc/init.d causes it to exit immediately with no messages. However, I tried starting transmission-daemon in foreground:

...:/tmp/home/root# transmission-daemon -g /opt/etc/transmission -f
transmission-daemon: symbol lookup error: /opt/lib/libdht.so: undefined symbol: dht_random_bytes

Checking the mentioned .so file shows the following:

...:/tmp/home/root# opkg search /opt/lib/libdht.so
libdht - 2022-04-27-11123089-1

Reinstalling libdht package does not help, version string hints that it is a bit dated anyway.

I strongly suspect that it's entware problem, libdht should be upgraded for transmission to work."

atkinsom, 2023-06-26:
Further comment from sad-spirit on SNBForums

"Now I'm actually sure of this, I managed to start transmission-daemon with the switch to disable DHT

 -O   --no-dht                          Disable distributed hash tables (DHT)

and successfully downloaded a torrent."

zyxmon, 2023-06-27:
What architecture (feed) has problems with new transmission? armv7sf-k2.6?

zyxmon, 2023-06-27:
Transmission 4.03 bug is discussed here - https://github.com/Entware/Entware/issues/953

zyxmon, 2023-06-27:
I am closing issue as incorrect. Entware uses the latest libdht version.

closed 2023-06-27 by zyxmon