Entware/Entware #953: transmission 4.03 does not start on arm7-k2.6

zyxmon, 2023-06-27:
The problem is discussed here - https://www.snbforums.com/threads/transmission-stopped-working.84082/post-849863
The temporary solution is to disable dht.

AkteX, 2023-08-12:
The root cause of the problem is that the function dht_random_bytes is not available in libdht. To solve this problem the source file dht-example.c must be compiled and linked to the library.

```
define Build/Compile
<------>$(TARGET_CC) $(FPIC) -Wall -c -o $(PKG_BUILD_DIR)/dht.o $(PKG_BUILD_DIR)/dht.c
<------>$(TARGET_CC) $(FPIC) -Wall -c -o $(PKG_BUILD_DIR)/dht-example.o $(PKG_BUILD_DIR)/dht-example.c
<------>$(TARGET_CC) -shared -lcrypt -o $(PKG_BUILD_DIR)/libdht.so $(PKG_BUILD_DIR)/dht.o $(PKG_BUILD_DIR)/dht-example.o
endef
```

zyxmon, 2023-08-12:
updated libdht package - https://test.entware.net/armv7sf-k2.6/libdht_2022-04-27-11123089-1a_armv7-2.6.ipk

AkteX, 2023-08-12:
The problem occurs with all architectures?. Can you please update the libdht package for all of them? I tested it with x86-2.6.

zyxmon, 2023-08-13:
It looks like only 2 architectures (k2.6) need this.

AkteX, 2023-08-13:
I checked bin.entware.net and found libdht in all architectures. As long as libdht is used it shall be build with dht-example.c. Otherwise some symbols which are exported by dht.h are not available. This was the reason why transmission didn't start.

zyxmon, 2023-08-13:
tr-dht.cc (transmission sources) has dht_random_bytes function. For some reasons it is not used in two architectures (for old kernels).
aarch64 transmission runs correctly with dht enabled and without dht_random_bytes in libdht. It also runs correctly with "new" дшивре (just tested it). I am waiting response from  arm7-k2.6 owners to commit your changes to the repo.

AkteX, 2023-08-13:
I just tested Entware arm7-k2.6. With libdht from the repo transmission-daemon didn't start. After the installation of https://test.entware.net/armv7sf-k2.6/libdht_2022-04-27-11123089-1a_armv7-2.6.ipk transmission-daemon is running.

zyxmon, 2023-08-14:
Fixed via https://github.com/Entware/entware-packages/commit/fb12474c80fd9834482c13ca549894d0b0ac4a0c

closed 2023-08-14 by zyxmon