Entware/Entware #954: Restic

newdanceradio, 2023-06-29:
Could it be possible to add "restic" package to entware repository.
Restic is very powerfull backup solution.

Project URL: https://restic.net/

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [aarch64-k3.10] and get a feedback about it.

zyxmon, 2023-07-13:
aarch64 package for testing - https://test.entware.net/aarch64-k3.10/restic-go_0.15.2-1_aarch64-3.10.ipk

newdanceradio, 2023-07-13:
Thank you @zyxmon 
I can confirm, that package is working properly
[image]

ryzhovau, 2023-09-07:
Thanks for feedback!

closed 2023-09-07 by ryzhovau