Entware/Entware #959: QPKG creates a symbolic link /opt

olluz, 2023-07-07:
**Describe the bug**
Not really a bug, but a change request. With every start of the QPKG (Entware.sh start) the script will delete /opt and create a symbolic link to the Entware directory:
# sym-link $QPKG_DIR to /opt
/bin/rm -rf /opt
/bin/ln -sf $QPKG_DIR /opt

This has the negative side effect that other symbolic links inside /opt (from other QPKGs) get deleted and won't work anymore.
Would it be possible to NOT delete directory/symbolic link with every start and instead link /opt/Entware to $QPKG_DIR ?
I hope this wouldn't require re-compiling all packages...

Thanks for all the hard work, this is greatly appreciated!

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]: x64-k3.2
 - Device [e.g. Asus RT-AX88U router]: QNAP TVS-h1288X
 - Firmware version [e.g. asuswrt-merlin]: QuTS hero h5.0.1.2376

zyxmon, 2023-07-07:
1. Qpkg is not a part of Entware.
2. You can use SortMyPackages package from OneCD to change this.
3. It is a bad idea to mix Entware with other Qnap packages that use /opt. /opt is hardcoded in Entware - just ask other qpkg makers not to use /opt.

zyxmon, 2023-07-07:
You can unpack and modify Entware qpkg and share it with other users.

closed 2023-07-07 by zyxmon

olluz, 2023-07-09:
> 1. Qpkg is not a part of Entware.
Yes, I'm aware of that. But you are the author of the QPKG, aren't you ?
> 2. You can use SortMyPackages package from OneCD to change this.
Sorry, I don't understand what SortMyPackages usage has to do with this issue. Even if I start Entware sooner or later it will still delete /opt and assume it's the only package using /opt
> 3. It is a bad idea to mix Entware with other Qnap packages that use /opt. /opt is hardcoded in Entware - just ask other qpkg makers not to use /opt.
Yes, it is. But not because of the others deleting /opt. Only Entware is deleting /opt and hence rendering the other packages unusable. 
I just think it is bad practice assuming that Entware owns /opt and therefore deleting and linking it to Entware only. What makes you think you are the only one that is allowed to use /opt and everybody else shouldn't ?
Of course I can ask every other QPKG developer to change their usage of /opt, although they did nothing wrong. I just wish you would at least think about you poor choice of design. Thanks

zyxmon, 2023-07-09:
Entware qpkg AFAIR was based on an official Optware QPKG. No other packages used /opt when Entware qpkg was made (but Optware). There are no rules even now what any qpkg can use and modify.

Here is info on SortMyPackages - https://forum.qnap.com/viewtopic.php?t=133132
(registration required). If you run it - it will make Entware start first and other qpkg using /opt will not be deleted.

olluz, 2023-07-09:
Thanks, I will have a look at SortMyPackages to overcome this problem.

Nevertheless, since even QNAP is using /opt it still is a bad idea to just remove the entire folder. I'm pretty sure they have been creating QPKGs before Entware qpkg was made.

You should at least have a warning in the qpkgs description that this will happen and to use SortyMyPackages to prevent it from happening.

Again, please consider not deleting folders that are commonly used by others as well.

According to the Linux foundation: /opt is “reserved for the installation of add-on application software packages.”