Entware/Entware #964: Tailscale — include iptables rules in package

Tokarak, 2023-07-25:
I had a painfully long session of localising the symptoms of this problem; but that's done now. 

Appending `/usr/sbin/iptables -I INPUT -i tailscale0 -m state --state NEW -j ACCEPT` to `/jffs/scripts/nat-start` solved it for me.

Now I'm wondering what `--netfliter-mode=on` is supposed to do; all evidence seems to point to doing exactly this. I can't find any technical documentation on it, so I can't be certain. https://openwrt.org/docs/guide-user/services/vpn/tailscale/start#openwrt_2203_issue has given me the most information — but I'm on asusWRT-merlin, and have iptables installed. In such a case, this issue should be migrated upstream to Tailscale.

If `--netfliter-mode` is going to stay broken, it would be nice if the rule was added to `S06tailscaled` (or something like that). If the Entware policy does not allow that, then maybe a caveats message somewhere in the docs.

zyxmon, 2023-07-25:
In most cases tailscale sets correct iptables rules in the system. If in your case some extra rules are needed please contact tailscale support - probably they will add them to the program.
I am running tailscale on qnap nas with no extra rules.

closed 2023-07-25 by zyxmon

Tokarak, 2023-07-25:
I finally worked it out. 
- The ts-input chain returns to the INPUT chain, where it is dropped by default as per asusWRT.
- asusWRT can rebuild iptables under config changes

Thank you. I'll move this issue upstream.