Entware/Entware #965: sstp-client version >1.0.15

dominiaz, 2023-07-26:
There is a bug with sstp-client 1.0.15 which is described here:

[https://sourceforge.net/p/sstp-client/discussion/1499218/thread/f96014d027/?limit=25](url)

sstp-client 1.0.15 is useless with pppd 2.4.8 in Entware repo.

1) Please compile sstp-client 1.0.15 with patch from the link 
or 
2) Compile new package with fixed issue - minimal version sstp-client 1.0.16 or newer.

I need that package in armv7sf-k3.2 architecture.

zyxmon, 2023-07-26:
here is it (patched) - https://test.entware.net/armv7sf-k3.2/sstp-client_1.0.15-1a_armv7-3.2.ipk
Please report results.

dominiaz, 2023-07-26:
Unfortunately there is still an issue.

root@p722:/ # ln -s /opt/lib/sstp-pppd-plugin.so /opt/lib/pppd/2.4.8/sstp-pppd-plugin.so
root@p722:/ # /opt/sbin/pppd call peer-sstp-example.txt
/opt/sbin/pppd: /opt/lib/pppd/2.4.8/sstp-pppd-plugin.so: undefined symbol: mppe_get_recv_key
/opt/sbin/pppd: Couldn't load plugin sstp-pppd-plugin.so

zyxmon, 2023-07-26:
https://test.entware.net/armv7sf-k3.2/sstp-client_1.0.16-1_armv7-3.2.ipk
and this one?

zyxmon, 2023-08-15:
fixed via https://github.com/Entware/entware-packages/commit/c832a8ce83cdde0fa4edb6f53a0c6f00266a7b0b

closed 2023-08-15 by zyxmon