Entware/Entware #967: Headscale

wellloaded, 2023-07-27:
**Describe the solution you'd like**
Heascale is an open source self-hosted version of Tilescale that is instead partially proprietary. It allows for modern VPN (VPS) to be created centrally over wireguard.

https://github.com/juanfont/headscale

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

zyxmon, 2023-08-12:
Makefile is ready. What arch apk can you test? I'll upload it.

wellloaded, 2023-08-15:
Thanks. As per OP I'm on build: armv7-2.6

zyxmon, 2023-08-15:
Here we are:
Normal - https://test.entware.net/armv7sf-k2.6/headscale_v0.22.3-1_armv7-2.6.ipk
No hard float - https://test.entware.net/armv7sf-k2.6/headscale_nohf_v0.22.3-1_armv7-2.6.ipk
Please report if the package is OK.

wellloaded, 2023-08-17:
The Normal version (might be wanted?) doesn't appear to behave in a "traditional" way

root@sparrow:/tmp# headscale --help
Illegal instruction
root@sparrow:/tmp# headscale -h
Illegal instruction
root@sparrow:/tmp# headscale apikeys create --expiration 90d
Illegal instruction
root@sparrow:/tmp# headscale
Illegal instruction

So if I remove:
root@sparrow:/tmp# opkg remove headscale
Removing package headscale from root...

and try to install the no hard float I get this:
root@sparrow:/tmp# opkg install /tmp/headscale_nohf_v0.22.3-1_armv7-2.6.ipk
Collected errors:
 * pkg_init_from_file: Malformed package file /tmp/headscale_nohf_v0.22.3-1_armv7-2.6.ipk.

zyxmon, 2023-08-17:
Links to work in opkg should be http, not https. My mistake!
```
~ # opkg install http://test.entware.net/armv7sf-k2.6/headscale_nohf_v0.22.3-1_armv7-2.6.ipk
Downloading http://test.entware.net/armv7sf-k2.6/headscale_nohf_v0.22.3-1_armv7-2.6.ipk
Installing headscale_nohf (v0.22.3-1) to root...
Configuring headscale_nohf.
~ # headscale --help

headscale is an open source implementation of the Tailscale control server

https://github.com/juanfont/headscale

Usage:
  headscale [command]

Available Commands:
  apikeys     Handle the Api keys in Headscale

```
You should always use nohf packages if available for your CPU.

wellloaded, 2023-08-17:
From a preliminary test, despite the fact that I couldn't set it up to be 100% operational yet, I didn't run into any obvious issue. Even the error messages in the config look fair and well reported on.

ryzhovau, 2023-09-07:
Closing for now.

closed 2023-09-07 by ryzhovau