Entware/Entware #968: [git] The 'git' protocol 'git://' is not supported

changbindu, 2023-08-05:
**Describe the bug**
It seems the git protocol 'git://' is not enabled, while this is a very basic protocol.

```
$ git clone git@github.com:util-linux/util-linux.git
Cloning into 'util-linux'...
/usr/bin/ssh: Ignoring unknown configuration option 'SendEnv=GIT_PROTOCOL'

/usr/bin/ssh: Connection to git@github.com:22 exited: No auth methods could be used.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

zyxmon, 2023-08-05:
You are cloning using ssh, not git. It looks like you have no rights.
Github recommends https protocol that works in Entware from scratch.

closed 2023-08-05 by zyxmon

changbindu, 2023-08-05:
> You are cloning using ssh, not git. It looks like you have no rights.
> Github recommends https protocol that works in Entware from scratch.

This is an open source repo, and public to veryone.