Entware/Entware #969: [LDAP] missing -dev package

bioinfornatics, 2023-08-09:
Dear,

Firstly thanks for your amazing works.
As I use both LDAP and automation through ansible I need to build the python package `python-ldap` which needs LDAP header file such as `lber.h`.

As you provides ldap_server package could you to add the ldap_server-dev package

Thanks
Best regards

zyxmon, 2023-08-09:
dev packages are not needed to build python modules - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

closed 2023-08-09 by zyxmon

bioinfornatics, 2023-08-09:
Dev packages are needs some time to build python module
below the reproducible issue, you wil found that as they are no openldap package, they are no way to get header files like `lber.h` which are needs to build the module

On QNAP with entware repo

```
# python -m venv test 
# type gcc
gcc is /opt/bin/gcc
# source test/bin/activate
(test) [~] # pip install python-ldap
Collecting python-ldap
  Using cached python-ldap-3.4.3.tar.gz (377 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyasn1>=0.3.7
  Using cached pyasn1-0.5.0-py2.py3-none-any.whl (83 kB)
Collecting pyasn1-modules>=0.1.5
  Using cached pyasn1_modules-0.3.0-py2.py3-none-any.whl (181 kB)
Building wheels for collected packages: python-ldap
  Building wheel for python-ldap (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for python-ldap (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [97 lines of output]
      /share/CACHEDEV1_DATA/.qpkg/Entware/tmp/pip-build-env-j9tog6ja/overlay/lib/python3.11/site-packages/setuptools/config/setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
      !!
      
              ********************************************************************************
              The license_file parameter is deprecated, use license_files instead.
      
              By 2023-Oct-30, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        parsed = self.parsers.get(option_name, lambda x: x)(value)
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-311
      copying Lib/ldapurl.py -> build/lib.linux-x86_64-cpython-311
      copying Lib/ldif.py -> build/lib.linux-x86_64-cpython-311
      creating build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/__init__.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/async.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/asyncsearch.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/cidict.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/compat.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/constants.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/dn.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/filter.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/functions.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/ldapobject.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/logger.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/modlist.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/pkginfo.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/resiter.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/sasl.py -> build/lib.linux-x86_64-cpython-311/ldap
      copying Lib/ldap/syncrepl.py -> build/lib.linux-x86_64-cpython-311/ldap
      creating build/lib.linux-x86_64-cpython-311/ldap/controls
      copying Lib/ldap/controls/__init__.py -> build/lib.linux-x86_64-cpython-311/ldap/controls
      copying Lib/ldap/controls/deref.py -> build/lib.linux-x86_64-cpython-311/ldap/controls
      copying Lib/ldap/controls/libldap.py -> build/lib.linux-x86_64-cpython-311/ldap/controls
      copying Lib/ldap/controls/openldap.py -> build/lib.linux-x86_64-cpython-311/ldap/controls
[... 55 lines cut ...]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for python-ldap
Failed to build python-ldap
ERROR: Could not build wheels for python-ldap, which is required to install pyproject.toml-based projects

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
```

On linux you will file the required files into -dev/-devel package

```
sudo dnf provides '*/lber.h'
Last metadata expiration check: 1:52:57 ago on Wed Aug  9 20:51:05 2023.
openldap-devel-2.6.4-1.fc38.i686 : LDAP development libraries and header files
Repo        : fedora
Matched from:
Filename    : /usr/include/lber.h

openldap-devel-2.6.4-1.fc38.x86_64 : LDAP development libraries and header files
Repo        : fedora
Matched from:
Filename    : /usr/include/lber.h
```

zyxmon, 2023-08-09:
Please read wiki - it has an instruction on installing headers.

bioinfornatics, 2023-08-09:
thanks for your reply.

I read those wiki pages:
1. https://github.com/Entware/Entware/wiki/
2. https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules
3. https://github.com/Entware/Entware/wiki/Compile-packages-from-sources

so the closest thing that could cover the needs it is to follow the page n°3, I have to get  source of ldap_server package

I also search from command line

```
# opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
# opkg upgrade
# opkg search ldap
```

So here it is strange search give no result while grep works better

```
# opkg list | grep -i ldap
apache-mod-ldap - 2.4.52-1a - LDAP authentication/authorization module for the Apache HTTP Server.
asterisk-res-config-ldap - 20.0.1-1 - LDAP realtime interface.
freeradius3-mod-ldap - 3_0_21-2a - LDAP Authentication
libaprutil-ldap - 1.6.1-7 - libaprutil - LDAP driver
libopenldap - 2.6.4-1 - OpenLDAP Software is an open source implementation of the Lightweight Directory Access Protocol (LDAP). This package contains the shared LDAP client libraries, needed by other programs.
lighttpd-mod-authn_ldap - 1.4.70-1 - LDAP-based authentication module
lmdb - 0.9.29-1 - LMDB is an ultra-fast, ultra-compact key-value embedded data store developed for the OpenLDAP Project.
openldap-server - 2.6.4-1 - This package contains server programs required to provide LDAP services.
openldap-utils - 2.6.4-1 - This package contains client programs required to access LDAP servers.
php8-mod-ldap - 8.2.6-1 - LDAP shared module
strongswan-mod-ldap - 5.9.10-5 - StrongSwan LDAP plugin
stunnel - 5.67-4 - Stunnel is a program that allows you to encrypt arbitrary TCP connections inside SSL (Secure Sockets Layer) available on both Unix and Windows. Stunnel can allow you to secure non-SSL aware daemons and protocols (like POP, IMAP, LDAP, etc) by having Stunnel provide the encryption, requiring no changes to the daemon's code.
```

openldap is installed
```
# opkg install libopenldap  openldap-server
Package libopenldap (2.6.4-1) installed in root is up to date.
Package openldap-server (2.6.4-1) installed in root is up to date.
```

But nor libopenldap or openldap-server provides needed hearder file
```
# opkg files libopenldap | grep '.h'
# opkg files openldap-server | grep '\.h'
#
```
Using google with `site:github.com entware installing headers` i found this old page: https://github.com/Entware/Entware-ng/wiki/Using-gcc-(native-compilation)
which is the same spirit of the page n°3

So to my understanding while you provides sometime `-dev` pakage as [libncurses-dev](https://bin.entware.net/x64-k3.2/libncurses-dev_6.4-2_x64-3.2.ipk) those are exceptions. End user have to go to the source code, overriding INCLUDE_PATH in order to tell to  gcc where to look at and try to build the python module.

zyxmon, 2023-08-10:
The first link on the first line on ref 2 you cite is - https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation