Entware/Entware #976: can not run dig in asuswrt merlin  386.12

sacc-leo, 2023-09-08:
**Describe the bug**
[package name]: bind-dig
**Environment 
 - Entware feed: [armv7sf-k2.6]
 - Device [Asus RT-AC66U_b1 router]
 - Firmware version [asuswrt-merlin ]

when i run dig in terminal, i got this message as blow:

netmgr/uverr2result.c:101:isc___nm_uverr2result(): unable to convert libuv error code in udp_send_cb (netmgr/udp.c:802) to isc_result: -38: function not implemented

zyxmon, 2023-09-20:
We have downgraded libuv for this feed. It should fix the bug

```
opkg update
opkg remove libuv  --force-depends
opkg install libuv
```

closed 2023-09-20 by zyxmon

ColinTaylorUK, 2024-10-23:
> Entware Feed: [armv7sf-k2.6] Device: [ASUS RT-AC68U] Firmware: [Merlin 386.14]
> 
> Having the option [Parallel use Entware-backports Repo] enabled in Entware settings, causes the script to still download the latest version of libuv (1.48.0-1) with the above reported bug
> 
> I had to disable the option to get the downgraded libuv (1.44.1-1)

This has nothing to do with the Entware devs. The "Parallel use..." option is part of a bespoke asuswrt-Merlin script (amtm). And the incorrect version of libuv is being obtained from an unofficial repository (maurerr.github.io/entware-armv7-k26/).

tarek-zy, 2024-10-23:
Thank you very much! I guess I will disable the parallel use feature for good in this old router. I already downgraded to the working versions available in the official repo.