Entware/Entware #977: Latest update (Sep 2023) leaves host command producing hardware error reports in syslog.log

original-birdman, 2023-09-08:
This is on a Asus RTAC86-U router, running Asus firmware 3.0.0.4.386_51915.

The package is:
bind-host - 9.18.16-1

When I run the host command it runs OK. Even if I `strace` it there is no sign of anything untoward. However, on each run a message like this get reported in syslog.
```
Sep  8 11:57:23 kernel: Code: aa0503e4 aa0603e5 aa0703e6 d4000001 (b13ffc1f)
Sep  8 11:57:23 kernel: CPU: 1 PID: 30056 Comm: host Tainted: P           O    4.1.27 #2
Sep  8 11:57:23 kernel: Hardware name: Broadcom-v8A (DT)
Sep  8 11:57:23 kernel: task: ffffffc0193db540 ti: ffffffc011bd0000 task.ti: ffffffc011bd0000
Sep  8 11:57:23 kernel: PC is at 0x7f7b8a5ee4
Sep  8 11:57:23 kernel: LR is at 0x7f7b7bd3a4
Sep  8 11:57:23 kernel: pc : [<0000007f7b8a5ee4>] lr : [<0000007f7b7bd3a4>] pstate: 20000000
Sep  8 11:57:23 kernel: sp : 0000007fddc96d10
Sep  8 11:57:23 kernel: x29: 0000007fddc96d10 x28: 0000000000000003
Sep  8 11:57:23 kernel: x27: 0000000000000000 x26: 0000000000000000
Sep  8 11:57:23 kernel: x25: 0000000000000000 x24: 0000000000000000
Sep  8 11:57:23 kernel: x23: 0000000000000100 x22: 0000000000000000
Sep  8 11:57:23 kernel: x21: 0000007f7b7db63c x20: 0000000006cd8d10
Sep  8 11:57:23 kernel: x19: 0000007f7b7db000 x18: 0000000000000000
Sep  8 11:57:23 kernel: x17: 000000000042a118 x16: 0000007f7bc6f340
Sep  8 11:57:23 kernel: x15: 081a171b01000000 x14: 000111c21c000000
Sep  8 11:57:23 kernel: x13: 0000000000000000 x12: 000000000000001f
Sep  8 11:57:23 kernel: x11: 0101010101010101 x10: 0000000000000006
Sep  8 11:57:23 kernel: x9 : 7f7f7f7f7f7fffff x8 : 00000000000001a9
Sep  8 11:57:23 kernel: x7 : 000000011b171a08 x6 : 000000011b171a08
Sep  8 11:57:23 kernel: x5 : 000000011b171a08 x4 : 081a171b01000000
Sep  8 11:57:23 kernel: x3 : 0000000000000000 x2 : 0000007f7b8a5ec0
Sep  8 11:57:23 kernel: x1 : 0000007fddc96d88 x0 : 0000000000000100
```

Oddly, if I copy back the bind-host - 9.18.11-3 version (and related libs, and use LD_LIBRARY_PATH to get it to run) it _also_ produces these messages. So the actual problem is a little deeper.

The 9.18.11-3 version did not produce these message when it was running on its real release.

original-birdman, 2023-09-08:
Actually, this may not be a hardware error report. But I thought it was when I first saw it.

original-birdman, 2023-09-10:
Tracked down the kernel source that produces this.

It must be a call to `dump_instr` (in traps.c) with arg1 == `""`.
So it's the one from `do_ni_syscall()`.

So it looks as though the latest Entware release is not compatible with kernel 4.1.27 syscalls.

Is this likely to change?

Is there any way for me to go back to the previous release?

ryzhovau, 2023-09-10:
I saw a similar [question on SNB](https://www.snbforums.com/threads/dig-can-not-run-in-386-12.86693/). Are you sure it's not about [dependency troubles](https://github.com/Entware/Entware/wiki#%EF%B8%8F-package-upgrade-notes)?

Please, reinstall Entware from scratch and provide the `strace /opt/bin/dig` log.

I've got no any asuswrt-merlin devices for now and can't reproduce this error on aarch64/x64 ones.

original-birdman, 2023-09-10:
There are no dependency troubles. The command actually _completes successfully_ - but reports the missing syscall to syslog along the way.

The SNB link was interesting though (thanks), especially this bit from jgarrigan:
```
I also encountered the same problem as [saccleo](https://www.snbforums.com/members/61064/). 
I resolved the issue by reverting to an older version of libuv.so.1.0.0. I had an older version on a USB stick.
```
I have daily backups, so also found the previous `libuv.so.1.0.0, and putting this in place _also_ resolves the issue _for me_.

So:
```
libuv - 1.44.1-1
```
is OK, but
```
libuv - 1.45.0-1
```
produces the syslog message.s

original-birdman, 2023-09-10:
Related, in so far as I tried to see how nslookup (== busybox) fared.

However, whereas `nslookup` claims that it can be run as:
```
BusyBox v1.36.1 (2023-06-03 17:45:19 UTC) multi-call binary.

Usage: nslookup HOST [DNS_SERVER]
```
in practice it _totally ignores_ the [DNS_SERVER] parameter and always does a lookup using the server mentioned in resolv.conf.
 
I have another nslookup/busybox combination (on my TV recorder):
```
BusyBox v1.36.0 () multi-call binary.

Usage: nslookup [-type=QUERY_TYPE] [-debug] HOST [DNS_SERVER]

Query DNS about HOST

QUERY_TYPE: soa,ns,a,aaaa,cname,mx,txt,ptr,srv,any
```
and that _does_ honour the [DNS_SERVER] parameter.

Should I open a separate issue for this?