Entware/Entware #981: node gets killed by SIGTRAP

ddokupil, 2023-09-13:
**Describe the bug**
[node-npm]: NodeJS gets Killed by --- SIGTRAP {si_signo=SIGTRAP, si_code=TRAP_BRKPT, si_addr=0x1560c44} ---

**Environment:**
 - Entware feed: ? Installed using amtm on asuswrt-merlin
 - Device: Asus RT-AX88U router
 - Firmware version asuswrt-merlin - 3004.388.4

I'm trying to run zigbee2mqtt on my router (https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing) but stuck on npm. Atfer enabling a swap file on my system I managed to successfully run npm install and npm ci but when I  try npm run build i get

```
admin@RT-AX88U-DB60:/tmp/mnt/toshiba/zigbee2mqtt# npm run build

> zigbee2mqtt@1.33.0 build
> tsc && node index.js writehash

#
# Fatal error in , line 0
# Check failed: page_allocator->DiscardSystemPages( reinterpret_cast<void*>(memory_area.begin()), memory_area.size()).
#
#
#
#FailureMessage Object: 0x7f8446dea0

#
# Fatal error in , line 0
# Check failed: page_allocator->DiscardSystemPages( reinterpret_cast<void*>(memory_area.begin()), memory_area.size()).
#
#
#
#FailureMessage Object: 0x7f7fffdde0

#
# Fatal error in , line 0
# Check failed: page_allocator->DiscardSystemPages( reinterpret_cast<void*>(memory_area.begin()), memory_area.size()).
#
#
#
#FailureMessage Object: 0x7f7fdfcea0

#
# Fatal error in , line 0
# Check failed: page_allocator->DiscardSystemPages( reinterpret_cast<void*>(memory_area.begin()), memory_area.size()). 1: 0xaee764  [npm run build]

#
#
#
#FailureMessage Object: 0x7fc840fe10
 1: 0xaee764  [npm run build]
 1: 0xaee764  [npm run build]
 2: 0x15585f8 V8_Fatal(char const*, ...) [npm run build]
 3: 0xe2c09c  [npm run build]
 4: 0xe54148 v8::internal::Sweeper::RawSweep(v8::internal::Page*, v8::internal::Sweeper::FreeListRebuildingMode, v8::internal::FreeSpaceTreatmentMode, v8::internal::Sweeper::SweepingMode, v8::base::LockGuard<v8::base::Mutex, (v8::base::NullBehavior)0> const&) [npm run build]
 1: 0xaee764  [npm run build]
 2: 0x15585f8 V8_Fatal(char const*, ...) [npm run build]
 3: 0xe2c09c  [npm run build]
 4: 0xe54148 v8::internal::Sweeper::RawSweep(v8::internal::Page*, v8::internal::Sweeper::FreeListRebuildingMode, v8::internal::FreeSpaceTreatmentMode, v8::internal::Sweeper::SweepingMode, v8::base::LockGuard<v8::base::Mutex, (v8::base::NullBehavior)0> const&) [npm run build]
 5: 0xe552f0 v8::internal::Sweeper::ParallelSweepPage(v8::internal::Page*, v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode) [npm run build]
 5: 0xe552f0 v8::internal::Sweeper::ParallelSweepPage(v8::internal::Page*, v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode) [npm run build]
 2: 0x15585f8 V8_Fatal(char const*, ...) [npm run build]
 2: 0x15585f8 V8_Fatal(char const*, ...) [npm run build]
 6: 0xe55cb0 v8::internal::Sweeper::ParallelSweepSpace(v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode, int, int) [npm run build]
 6: 0xe55cb0 v8::internal::Sweeper::ParallelSweepSpace(v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode, int, int) [npm run build]
 3: 0xe2c09c  [npm run build]
 3: 0xe2c09c  [npm run build]
 7: 0xe3d0c0 v8::internal::PagedSpace::ContributeToSweepingMain(int, int, int, v8::internal::AllocationOrigin) [npm run build]
 7: 0xe3d0c0 v8::internal::PagedSpace::ContributeToSweepingMain(int, int, int, v8::internal::AllocationOrigin) [npm run build]
 8: 0xe3d190 v8::internal::PagedSpace::RawRefillLabMain(int, v8::internal::AllocationOrigin) [npm run build]
 4: 0xe54148 v8::internal::Sweeper::RawSweep(v8::internal::Page*, v8::internal::Sweeper::FreeListRebuildingMode, v8::internal::FreeSpaceTreatmentMode, v8::internal::Sweeper::SweepingMode, v8::base::LockGuard<v8::base::Mutex, (v8::base::NullBehavior)0> const&) [npm run build]
 8: 0xe3d190 v8::internal::PagedSpace::RawRefillLabMain(int, v8::internal::AllocationOrigin) [npm run build]
[... 99 lines cut ...]
19: 0xe0822c v8::internal::MarkCompactCollector::EvacuatePagesInParallel() [node]
20: 0xe2402c v8::internal::MarkCompactCollector::Evacuate() [node]
21: 0xe24a5c v8::internal::MarkCompactCollector::CollectGarbage() [node]
22: 0xddf170 v8::internal::Heap::MarkCompact() [node]
23: 0xddfcac v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::internal::GarbageCollectionReason, char const*, v8::GCCallbackFlags) [node]
24: 0xde0250 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node]
25: 0xde1420 v8::internal::Heap::FinalizeIncrementalMarkingIfComplete(v8::internal::GarbageCollectionReason) [node]
26: 0xde433c v8::internal::IncrementalMarkingJob::Task::Step(v8::internal::Heap*) [node]
27: 0xde444c v8::internal::IncrementalMarkingJob::Task::RunInternal() [node]
28: 0xcef154 non-virtual thunk to v8::internal::CancelableTask::Run() [node]
29: 0xaeefa8 node::PerIsolatePlatformData::RunForegroundTask(std::unique_ptr<v8::Task, std::default_delete<v8::Task> >) [node]
30: 0xaf0acc node::PerIsolatePlatformData::FlushForegroundTasksInternal() [node]
31: 0x7f8c6b084c  [/opt/lib/libuv.so.1]
32: 0x7f8c6c3084  [/opt/lib/libuv.so.1]
33: 0x7f8c6b1614 uv_run [/opt/lib/libuv.so.1]
34: 0x9eeb28 node::SpinEventLoop(node::Environment*) [node]
35: 0xaccdd4  [node]
36: 0xacda28 node::NodeMainInstance::Run() [node]
37: 0xa5a658 node::LoadSnapshotDataAndRun(node::SnapshotData const**, node::InitializationResult const*) [node]
38: 0xa5c1fc node::Start(int, char**) [node]
39: 0x7f8bd3efb4 __libc_start_main [/opt/lib/libc.so.6]
40: 0x9ed578  [node]
Trace/breakpoint trap
admin@RT-AX88U-DB60:/tmp/mnt/toshiba/zigbee2mqtt#
```

asaddon, 2024-07-23:
Same issue on AX86U
Merlin version: [3004.388.8]

Is there any update/fix on this issue?

```
Welcome to Node.js v18.18.2.
Type ".help" for more information.
> 

#
# Fatal error in , line 0
# Check failed: page_allocator->DiscardSystemPages( reinterpret_cast<void*>(memory_area.begin()), memory_area.size()).
#
#
#
#FailureMessage Object: 0x7f88434de0

#
# Fatal error in , line 0
# Check failed: page_allocator->DiscardSystemPages( reinterpret_cast<void*>(memory_area.begin()), memory_area.size()).
#
#
#
#FailureMessage Object: 0x7fd0cb8b50
 1: 0xaf1154  [node]
 1: 0xaf1154  [node]
 2: 0x155b6f8 V8_Fatal(char const*, ...) [node]
 2: 0x155b6f8 V8_Fatal(char const*, ...) [node]
 3: 0xe2f03c  [node]
 3: 0xe2f03c  [node]
 4: 0xe570e8 v8::internal::Sweeper::RawSweep(v8::internal::Page*, v8::internal::Sweeper::FreeListRebuildingMode, v8::internal::FreeSpaceTreatmentMode, v8::internal::Sweeper::SweepingMode, v8::base::LockGuard<v8::base::Mutex, (v8::base::NullBehavior)0> const&) [node]
 4: 0xe570e8 v8::internal::Sweeper::RawSweep(v8::internal::Page*, v8::internal::Sweeper::FreeListRebuildingMode, v8::internal::FreeSpaceTreatmentMode, v8::internal::Sweeper::SweepingMode, v8::base::LockGuard<v8::base::Mutex, (v8::base::NullBehavior)0> const&) [node]
 5: 0xe58290 v8::internal::Sweeper::ParallelSweepPage(v8::internal::Page*, v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode) [node]
 5: 0xe58290 v8::internal::Sweeper::ParallelSweepPage(v8::internal::Page*, v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode) [node]
 6: 0xe58c50 v8::internal::Sweeper::ParallelSweepSpace(v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode, int, int) [node]
 6: 0xe58c50 v8::internal::Sweeper::ParallelSweepSpace(v8::internal::AllocationSpace, v8::internal::Sweeper::SweepingMode, int, int) [node]
 7: 0xe40060 v8::internal::PagedSpace::ContributeToSweepingMain(int, int, int, v8::internal::AllocationOrigin) [node]
 7: 0xe40060 v8::internal::PagedSpace::ContributeToSweepingMain(int, int, int, v8::internal::AllocationOrigin) [node]
 8: 0xe40130 v8::internal::PagedSpace::RawRefillLabMain(int, v8::internal::AllocationOrigin) [node]
 8: 0xe40130 v8::internal::PagedSpace::RawRefillLabMain(int, v8::internal::AllocationOrigin) [node]
 9: 0xe27e08 v8::internal::EvacuationAllocator::Allocate(v8::internal::AllocationSpace, int, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node]
 9: 0xe27e08 v8::internal::EvacuationAllocator::Allocate(v8::internal::AllocationSpace, int, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node]
10: 0xe28014 v8::internal::EvacuateVisitorBase::TryEvacuateObject(v8::internal::AllocationSpace, v8::internal::HeapObject, int, v8::internal::HeapObject*) [node]
10: 0xe28014 v8::internal::EvacuateVisitorBase::TryEvacuateObject(v8::internal::AllocationSpace, v8::internal::HeapObject, int, v8::internal::HeapObject*) [node]
11: 0xe284e0 v8::internal::EvacuateNewSpaceVisitor::Visit(v8::internal::HeapObject, int) [node]
11: 0xe284e0 v8::internal::EvacuateNewSpaceVisitor::Visit(v8::internal::HeapObject, int) [node]
12: 0xe2895c void v8::internal::LiveObjectVisitor::VisitBlackObjectsNoFail<v8::internal::EvacuateNewSpaceVisitor, v8::internal::MajorNonAtomicMarkingState>(v8::internal::MemoryChunk*, v8::internal::MajorNonAtomicMarkingState*, v8::internal::EvacuateNewSpaceVisitor*, v8::internal::LiveObjectVisitor::IterationMode) [node]
12: 0xe2895c void v8::internal::LiveObjectVisitor::VisitBlackObjectsNoFail<v8::internal::EvacuateNewSpaceVisitor, v8::internal::MajorNonAtomicMarkingState>(v8::internal::MemoryChunk*, v8::internal::MajorNonAtomicMarkingState*, v8::internal::EvacuateNewSpaceVisitor*, v8::internal::LiveObjectVisitor::IterationMode) [node]
13: 0xe29730 v8::internal::FullEvacuator::RawEvacuatePage(v8::internal::MemoryChunk*, long*) [node]
13: 0xe29730 v8::internal::FullEvacuator::RawEvacuatePage(v8::internal::MemoryChunk*, long*) [node]
14: 0xdf6228 v8::internal::Evacuator::EvacuatePage(v8::internal::MemoryChunk*) [node]
14: 0xdf6228 v8::internal::Evacuator::EvacuatePage(v8::internal::MemoryChunk*) [node]
15: 0xdf683c v8::internal::PageEvacuationJob::Run(v8::JobDelegate*) [node]
15: 0xdf65d0 v8::internal::PageEvacuationJob::Run(v8::JobDelegate*) [node]
16: 0x14df4f8 v8::platform::DefaultJobWorker::Run() [node]
16: 0x14e78a0 v8::platform::DefaultJobState::Join() [node]
17: 0xaf29dc  [node]
18: 0x7f887c5c38  [/opt/lib/libpthread.so.0]
17: 0x14e78f8 v8::platform::DefaultJobHandle::Join() [node]
19: 0x7f8871d94c  [/opt/lib/libc.so.6]
18: 0xe0ad9c unsigned long v8::internal::MarkCompactCollectorBase::CreateAndExecuteEvacuationTasks<v8::internal::FullEvacuator, v8::internal::MarkCompactCollector>(v8::internal::MarkCompactCollector*, std::vector<std::pair<v8::internal::ParallelWorkItem, v8::internal::MemoryChunk*>, std::allocator<std::pair<v8::internal::ParallelWorkItem, v8::internal::MemoryChunk*> > >, v8::internal::MigrationObserver*) [node]
19: 0xe0b1cc v8::internal::MarkCompactCollector::EvacuatePagesInParallel() [node]
20: 0xe26fcc v8::internal::MarkCompactCollector::Evacuate() [node]
21: 0xe279fc v8::internal::MarkCompactCollector::CollectGarbage() [node]
22: 0xde2110 v8::internal::Heap::MarkCompact() [node]
23: 0xde2c4c v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::internal::GarbageCollectionReason, char const*, v8::GCCallbackFlags) [node]
[... cut ...]
Jul 24 03:05:54 kernel: x15: 0000007f6c000180 x14: 0000000000000014 
Jul 24 03:05:54 kernel: x13: 5d362e6f732e6362 x12: 696c2f62696c2f74 
Jul 24 03:05:54 kernel: x11: 0000007f88434410 x10: 0000007f88434410 
Jul 24 03:05:54 kernel: x9 : 00000000ffffffff x8 : 0000000000000040 
Jul 24 03:05:54 kernel: x7 : 2e6362696c2f6269 x6 : 0000000000000000 
Jul 24 03:05:54 kernel: x5 : 0000000000000007 x4 : 0000000000000000 
Jul 24 03:05:54 kernel: x3 : 0000000000000000 x2 : 0000000000000001 
Jul 24 03:05:54 kernel: x1 : 0000000000000000 x0 : 0000000000000001 
```

ryzhovau, 2025-01-16:
It looks like normal OOM behavior. Not sure router is the best place for node app building.

closed 2025-01-16 by ryzhovau