Entware/Entware #986: bin.entware.net using http protocol hang forever

26fe, 2023-09-23:
I am installing for the first time entware on a qnap TS-351 (x64). The installer hang forever. 

```
[/share/homes/admin] # ps -eaf | grep opkg
  941 admin       936 S   grep opkg
10517 admin      4536 S   /usr/bin/wget http://bin.entware.net/x64-k3.2/installer/opkg -qO /opt/bin/opkg

```
Using the https work, but installer script use http. 

```
[/share/homes/admin] # /usr/bin/wget https://bin.entware.net/x64-k3.2/installer/opkg
--2023-09-23 11:09:03--  https://bin.entware.net/x64-k3.2/installer/opkg
Resolving bin.entware.net... 158.160.45.54
Connecting to bin.entware.net|158.160.45.54|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1155888 (1.1M) [application/octet-stream]
Saving to: ‘opkg’

opkg                          100%[==============================================>]   1.10M  2.42MB/s    in 0.5s    

2023-09-23 11:09:04 (2.42 MB/s) - ‘opkg’ saved [1155888/1155888]

```

Please reconfigure bin.entware.net to correct redirect http requesto to https.
Thank you

zyxmon, 2023-09-23:
Entware uses http because a lot of devices that use Entware do not support downloading through https.
This was discussed many times.
Fix your internet connection or use another one.

zyxmon, 2023-09-23:
```
/share/CACHEDEV1_DATA/.qpkg/Entware/root # /usr/bin/wget http://bin.entware.net/x64-k3.2/installer/opkg -qO /opt/bin/opkg2
/share/CACHEDEV1_DATA/.qpkg/Entware/root # ls /opt/bin/opk*
/opt/bin/opkg   /opt/bin/opkg2
/share/CACHEDEV1_DATA/.qpkg/Entware/root #

```
qnap ts-253 (x64)

closed 2023-09-23 by zyxmon