Entware/Entware #989: avahi busted on asuswrt-merlin

privacyguy123, 2023-10-02:
**Describe the bug**
Same as https://github.com/Entware/Entware-ng/issues/376 but I didn't want to necrobump. I don't understand how to fix it ... ELI5

**Environment (please complete the following information):**
 - Entware feed: ?
 - Device RT-AX58U
 - Firmware version asuswrt-merlin  3004.388.4

zyxmon, 2023-10-02:
Just create nogroup group on your device. This is device specific.

closed 2023-10-02 by zyxmon

privacyguy123, 2023-10-02:
> Just create nogroup group on your device. This is device specific.

I've tried that and it's still busted. All avahi-tools complain the daemon isn't running. It is.

More stuff that makes no sense to me

```
RT-AX58U-5468:/tmp/home/root# strings `which avahi-daemon` | grep run/
/opt/var/run/avahi-daemon//pid
/opt/var/run/avahi-daemon/
mkdir("/opt/var/run/avahi-daemon/"): %s
Failed to create runtime directory /opt/var/run/avahi-daemon/.
/opt/var/run/avahi-daemon/socket
```

privacyguy123, 2023-10-03:
Sorry this isn't a sufficient answer.

How can I get avahi to recognize it's running as a daemon on asuswt-merlin? I've read your dbus dependency package is expecting to run as root and that user is not available to me, I can't even change my log in name to that.

zyxmon, 2023-10-03:
The best place to ask your questions is - https://www.snbforums.com/forums/asuswrt-merlin-addons.60/
We do not provide help for specific devices. AFAIK - no one from Entware team has asuswrt-merlin router.

privacyguy123, 2023-10-04:
Nobody over there knows why it doesn't work either.