Entware/Entware #991: Can you compile an python3-orjson? Thank you

fwebcc, 2023-10-11:
Can you compile an python3-orjson? Thank you
Because homeassistant needs

zyxmon, 2023-10-11:
Have you tried to build/install it yourself as described here - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

fwebcc, 2023-10-13:
I tried and needed 
Downloading orjson-3.9.9.tar.gz (4.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.7/4.7 MB 128.4 kB/s eta 0:00:00
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [57 lines of output]
      Collecting maturin<2,>=1
        Using cached maturin-1.3.0.tar.gz (175 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Building wheels for collected packages: maturin
        Building wheel for maturin (pyproject.toml): started
        Building wheel for maturin (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Building wheel for maturin (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [35 lines of output]
            /tmp/mnt/fwebcc/tmp/pip-build-env-eziw7y08/overlay/lib/python3.11/site-packages/setuptools/config/_apply_pyprojecttoml.py:80: SetuptoolsWarning: `install_requires` overwritten in `pyproject.toml` (dependencies)
              corresp(dist, value, root_dir)
            running bdist_wheel
            running build
            running build_py
            creating build
            creating build/lib.linux-armv7l-cpython-311
            creating build/lib.linux-armv7l-cpython-311/maturin
            copying maturin/import_hook.py -> build/lib.linux-armv7l-cpython-311/maturin
            copying maturin/__main__.py -> build/lib.linux-armv7l-cpython-311/maturin
            copying maturin/__init__.py -> build/lib.linux-armv7l-cpython-311/maturin
            running egg_info
            creating maturin.egg-info
            writing maturin.egg-info/PKG-INFO
            writing dependency_links to maturin.egg-info/dependency_links.txt
            writing requirements to maturin.egg-info/requires.txt
            writing top-level names to maturin.egg-info/top_level.txt
            writing manifest file 'maturin.egg-info/SOURCES.txt'
            reading manifest file 'maturin.egg-info/SOURCES.txt'
            reading manifest template 'MANIFEST.in'
            warning: no files found matching '*.json' under directory 'src/python_interpreter'
            writing manifest file 'maturin.egg-info/SOURCES.txt'
            running build_ext
            running build_rust
            error: can't find Rust compiler
      
            If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
      
            To update pip, run:
      
                pip install --upgrade pip
      
            and then retry package installation.
      
            If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for maturin
      Failed to build maturin
      ERROR: Could not build wheels for maturin, which is required to install pyproject.toml-based projects
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

zyxmon, 2023-10-23:
Entware does not support Rust now.
This package cannot be built (crosscompiled or native compilation).

closed 2023-10-23 by zyxmon

ryzhovau, 2024-02-22:
Rust packages has been added, including `python3-orjson`. Binaries will arrive soon.