Entware/Entware #994: [Package Request] corosync-qdevice and corosync-qnetd

cwiggs, 2023-10-22:
**Describe the solution you'd like**
I would like to use my asuswrt router as a qdevice in a proxmox cluster.  In order to use the router as a qdevice we need 2 packages: corosync-qdevice and corosync-qnetd`, along with 2 dependencies NSS, and corosync.

Here is a link to the code repo: https://github.com/corosync/corosync-qdevice

I've installed corosync-qdevice on multiple raspberry pi's so I assume there are arm packages that will work.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ylavi, 2023-10-23:
I just came to make this request too!

I would be prepared to test on the mipssf-k3.4 (?) platform (I haven't yet installed DD-WRT and Entware on my device, but as far as I know it has a mips_24kc processor).

ptath, 2024-05-12:
me too, arm64

closed 2025-08-08 by The-BB