Entware/Entware #999: ds 220+

maciej1993, 2023-11-04:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 
what is the appropriate repository for ds 220+

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - Device [e.g. Asus RT-AX88U router]
 - Firmware version [e.g. asuswrt-merlin]

krvi, 2023-11-05:
Synology DS220+ uses an Intel Celeron J4025 which. as far as I can tell, has the `x86-64` instruction set. Therefore you would use the `x64-k3.2` feed.

zyxmon, 2023-11-05:
This is a place to report bugs, not for support. Use qnap forum please - https://forum.qnap.com/viewforum.php?f=351 (you must log in to view it).

closed 2023-11-05 by zyxmon