Entware/Entware #1007: [Package Request]

superspost, 2023-12-19:
libxcb-cursor0

- [x ] No similar package already added and/or this one got some new killer feature(s),
- [ x] Package is in active development (at least, not abandoned for 5 years or more),
- [ x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2023-12-20:
Please, tell us what are you trying to implement? 
We add any libraries only as dependencies for applications.

superspost, 2023-12-20:
I have a Synology DS1815+ and ran into trouble installing the latest Calibre docker container. Searching for a solution, it appears that the kernel of the NAS might be too old and it was suggested that installing this library on the host would solve the problem.

I have succeeded to install Entware, but do not know how to install this library.

Sorry if this is the wrong place to make this request.

superspost, 2023-12-20:
sorry i forgot to mention the link to the thread that suggested the solution. 

[link to thread](https://github.com/linuxserver/docker-calibre/issues/128)

ryzhovau, 2024-02-24:
We’re not going to add Calibre or some package that depends on libxcb-cursor0, sorry.

closed 2024-02-24 by ryzhovau