Entware/Entware #1021: Entware Bus Error

ChrisLipscombe, 2024-02-25:
Trying to install Entware while rooting Creality K1 3d printer.  Get bus errors.  Prevents ffmpeg from being installed when I try to install the timelapse feature.

 Installing Entware...

Making /opt directory on data partition where there is space, and adding a symbolic link...
Downloading generic installer script...
Connecting to raw.githubusercontent.com (185.199.108.133:443)
writing to stdout
-                    100% |*****************************************************************************************|  3222  0:00:00 ETA
written to stdout
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
Info: Basic packages installation...
Bus error
Bus error
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to $PATH variable
Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
Adding /opt/bin and /opt/sbin to the start of the PATH in the system profile...
Adding startup script...

zyxmon, 2024-02-25:
I am afraid entware team cannot help you. You are using instruction and scripts from here - https://github.com/Guilouz/Creality-K1-and-K1-Max/ - so all the installation questions should go there.

Guilouz, 2024-02-25:
It's looks like there is an issue with latest build of opkg released on February 24: http://bin.entware.net/aarch64-k3.10/installer/

/opt/bin/opkg update
/opt/bin/opkg upgrade

Have stopped to work.

```
Making /opt directory on data partition where there is space, and adding a symbolic link...
Running generic installer script...
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
Info: Basic packages installation...
Bus error
Bus error
```

zyxmon, 2024-02-25:
opkg was not modified for 9 months an was not changed on February  - https://github.com/Entware/Entware/tree/master/package/system/opkg

Guilouz, 2024-02-25:
@zyxmon 

[image]

ryzhovau, 2024-02-25:
Can't reproduce that on Keenetic Router:
```
~ # cd /opt/bin/
/opt/bin # mv opkg opkg.bak
/opt/bin # wget -q http://bin.entware.net/aarch64-k3.10/installer/opkg
/opt/bin # chmod +x ./opkg
/opt/bin # ./opkg --version
opkg version d038e5b6d155784575f62a66a8bb7e874173e92e (2022-02-24)
```

Guilouz, 2024-02-25:
@ryzhovau My bad, i'm not use aarch version, i use mipselsf version.

It's look like not updated since May 2023 but not working, it worked before.

When I use previous version it's working, don't really understand what's happens with current version.

ryzhovau, 2024-02-25:
Looks like some syscall is missed in kernel. Just checked mipsel version on ancient Padavan frimware from 2016, looks fine:
```
/opt/home/admin # opkg --version
opkg version d038e5b6d155784575f62a66a8bb7e874173e92e (2022-02-24)
/opt/home/admin # uname -a
Linux WT3020H 3.4.113 #1 Wed Oct 16 13:35:50 MSK 2019 mips GNU/Linux
```

Guilouz, 2024-02-25:
@ryzhovau Could this be related to these updated files?

[image]

Nothing has changed on the OS, it's a 3D printer (no firmware change, no kernel change).
Installer worked before until February 23

Guilouz, 2024-02-25:
@ryzhovau 
[image]

Last error is normal, I have just downloaded opkg.

Guilouz, 2024-02-25:
Installer install entware-opt package released on February 24: entware-opt_227000-3_all.ipk

It also replace opkg and it stop to work, I have Dbus error with many of my packages.
Previous version was working before, how i can install previous version ?

dwaite, 2024-02-26:
I believe this is related to https://github.com/Entware/Entware/commit/8ad1c9e99361c675b609425b1af00e38b7bf05eb , as the Ingenic CPU on this board does not support MIPS16.

Guilouz, 2024-02-26:
Yes it use Ingenic x2000E / xburst II Soc

@ryzhovau is it possible to restore this, a big community use it on many Creality printers.

ChrisLipscombe, 2024-02-26:
I have been following this thread, but everything is over my head. I hope whatever happened will be fixed. Like Guilouz said, a large community use this on Creality Printers, myself included.

Kev608, 2024-02-26:
I also have this problem.

``` Installing Entware...

Making /opt directory on data partition where there is space, and adding a symbolic link...
Downloading generic installer script...
Connecting to raw.githubusercontent.com (185.199.111.133:443)
writing to stdout
-                      0% |                                                                                                                                                                                                                      |     0 --:--:-- ETAInfo: Checking for prerequisites and creating folders...
-                    100% |**********************************************************************************************************************************************************************************************************************|  3222  0:00:00 ETA
written to stdout
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
Info: Basic packages installation...
Bus error
Bus error
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to $PATH variable
Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
Adding /opt/bin and /opt/sbin to the start of the PATH in the system profile...
Adding startup script...

 ✓ Entware has been installed successfully!```

ryzhovau, 2024-02-26:
Please, try to install Entware from http://test.entware.net/mipselsf-k3.4.

bcm63xx is the only OpenWrt MIPS32 target without MIPS16 extension. Not sure I want to turn it off MIPS16 in favor of few rare devices, sorry. 

I'll keep this issue open for a while to reveal similar caces.

Guilouz, 2024-02-26:
@ryzhovau it's working ! Is it possible to keep it ?

ryzhovau, 2024-02-27:
I’ll keep it for a week, so you'll be able to make your own mirror.