Entware/Entware #1022: opkg_install_cmd: Cannot install package

fireqpeg, 2024-02-25:
Hello! There is a rather unpleasant problem. With the new packages. rTorrent/rUtorrent plugins do not work, they do not install normally. It says some collected cannot install error. I say ok, I'll try another torrent client. qBittorrent doesn't work either, at least it started, but the basic username and password. It would not be appropriate admin/adminadmin. So, unfortunately, this is all very disappointing for me. :(

installed packages: htop mc nano rtorrent-easy-install rutorrent-plugin-erasedata rutorrent-plugin-tracklabels rutorrent-plugin-diskspace

I tried force depend too. opkg update, opkg uprgade before installation and afterwards. but then it doesn't work either. It's interesting that the previous September packages worked, but this new one doesn't.

I hope you will be able to solve the errors as soon as possible. Thanks in advance! Sincerely, Fireqpeg

Platform: http://bin.entware.net/mipselsf-k3.4
Padavan firmware: Asus rt-n56u

ryzhovau, 2024-02-25:
Can't reproduce it on x64/aarh64. Try to install Entware ftom scratch.

fireqpeg, 2024-02-26:
Hi ryzhovau!

So what happened was that I had fun with the installation. And the first one has the errors. If I start the installation a second time, everything is fine. It also uploads all the previous ones where you wrote an error. So with this solution it seems good to me. It has been running flawlessly for a day since yesterday. I would list the following packages: opkg install htop mc nano rtorrent-easy-install rutorrent-plugin-erasedata rutorrent-plugin-tracklabels rutorrent-plugin-diskspace. That's all, nothing more.

I hope I could help? There's a little glitch, but that's how it works.

Sincerely, fireqpeg

ryzhovau, 2024-08-08:
I just caught the same glitch and can't reproduce it again. So I'll keep this issue open for further investigation.

ryzhovau, 2025-05-21:
No chance to reproduce it again, closing for now.

closed 2025-05-21 by ryzhovau