Entware/Entware #1023: qbittorrent 4.6.3 web ui lock

xboxones1, 2024-02-26:
After updating to version 4.6.3, the username and password are not accepted in the web interface (by default admin:adminadmin). I reinstalled, deleted the folder with the configuration files, it doesn’t work. I installed the old version (4.5.4) from the archive, it works.

Entware feed: [e.g. aarch64]

ryzhovau, 2024-02-26:
Please, take a look at [qBittorrent changelog](https://qbittorrent.github.io/qBittorrent-website/news#mon-nov-20th-2023---qbittorrent-v4.6.1-release):

> The most important change has to do with WebUI. It affects users that haven't set their own credentials and are use the default ones…

xboxones1, 2024-02-26:
thanks, it worked