Entware/Entware #1024: The fossil version from last package is 2.22. Waiting version 2.23

oupfiz5, 2024-02-26:
**Describe the bug**
[fossil]: Install last fossil version from [rtndev ](https://github.com/Entware/rtndev/tree/master/fossil) commit [e7401fc459615bb9947a2307ec981ba79d03c121](https://github.com/Entware/rtndev/commit/e7401fc459615bb9947a2307ec981ba79d03c121). But it is version 2.22. 

Step buy step:
``` sh
opkg update
opkg upgade
opkg install fossil
fossil version
```

In my case:
``` sh
opkg info fossil
<--
Package: fossil
Version: 2.23-1
Depends: libc, libssp, librt, libpthread, libopenssl, libsqlite3, tcl
Status: install user installed
Section: net
Architecture: aarch64-3.10
Size: 2191943
Filename: fossil_2.23-1_aarch64-3.10.ipk
Description: Fossil is a self-contained, high-reliability, distributed software
 configuration management system, including Bug Tracking, Wiki, Forum,
 and Technotes.
Installed-Time: 1708953784
<--

fossil version
This is fossil version 2.22 [66ee0beb9b] 2023-05-31 15:26:08 UTC
```

Accordingly changes - waiting version **2.23**

**Environment (please complete the following information):**
 - Entware feed:  aarch64-k3.10
 - Device Asus RT-AX88U router
 - Firmware version - asuswrt-merlin

PS The binary program from http://bin.entware.net/aarch64-k3.10/fossil_2.23-1_aarch64-3.10.ipk contains fossil version 2.22.

FYI: The last fossil version is 2.24

oupfiz5, 2024-09-08:
the same problem - fossil is old. 
```
/tmp/etc# opkg install fossil
Installing fossil (2.24-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/fossil_2.24-1_aarch64-3.10.ipk
Configuring fossil.
/tmp/etc# fossil version
This is fossil version 2.22 [66ee0beb9b] 2023-05-31 15:26:08 UTC
```

ryzhovau, 2024-11-23:
Merged, thanks.

ryzhovau, 2025-01-16:
2.25 is here. Please, provide feedback.

closed 2025-01-16 by ryzhovau

oupfiz5, 2025-01-16:
> 2.25 is here. Please, provide feedback.

Updated. The version is 2.25. Thanks.