Entware/Entware #1026: Node shows "Illegal instruction" on MT7621 with mipselsf-k3.4

ximiximi1, 2024-03-04:
**Describe the bug**
[node]:
When run "node -v" it works fine, but it shows "Illegal instruction" for all other node cmd.

**Environment (please complete the following information):**
 - mipselsf-k3.4
 - MT7621. It was named hiwifi_hc5962 on openwrt
 - openwrt 23.05.2 with kernel 5.15.137
 
But node works fine on my another MT7621 device. The device is Xiaomi R3P , use kernel 3.10.14.
I have no idea about what make difference and cause the different result. It seems that the only difference is the kernel version, 5.15vs 3.10

Thanks.

ximiximi1, 2024-03-04:
**This is the cpuinfo.**
system type		: MediaTek MT7621 ver:1 eco:3
machine			: HiWiFi HC5962
processor		: 0
cpu model		: MIPS 1004Kc V2.15
BogoMIPS		: 586.13
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp mt
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit cdmm perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VPE			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

processor		: 1
cpu model		: MIPS 1004Kc V2.15
BogoMIPS		: 586.13
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp mt
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit cdmm perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VPE			: 1
VCED exceptions		: not available
VCEI exceptions		: not available

processor		: 2
cpu model		: MIPS 1004Kc V2.15
BogoMIPS		: 586.13
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp mt
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit cdmm perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 1
VPE			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

processor		: 3
cpu model		: MIPS 1004Kc V2.15
BogoMIPS		: 586.13
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp mt
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit cdmm perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 1
VPE			: 1
VCED exceptions		: not available
VCEI exceptions		: not available

**Below is the cpuinfo of the device that works well .**
system type             : MT7621
machine                 : Unknown
processor               : 0
cpu model               : MIPS 1004Kc V2.15
BogoMIPS                : 584.90
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp mt
shadow register sets    : 1
kscratch registers      : 0
core                    : 0
VPE                     : 0
VCED exceptions         : not available
VCEI exceptions         : not available

processor               : 1
cpu model               : MIPS 1004Kc V2.15
BogoMIPS                : 584.90
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp mt
shadow register sets    : 1
kscratch registers      : 0
core                    : 0
VPE                     : 1
VCED exceptions         : not available
VCEI exceptions         : not available

processor               : 2
cpu model               : MIPS 1004Kc V2.15
BogoMIPS                : 584.90
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp mt
shadow register sets    : 1
kscratch registers      : 0
core                    : 1
VPE                     : 0
VCED exceptions         : not available
VCEI exceptions         : not available

processor               : 3
cpu model               : MIPS 1004Kc V2.15
BogoMIPS                : 584.90
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp mt
shadow register sets    : 1
kscratch registers      : 0
core                    : 1
VPE                     : 1
VCED exceptions         : not available
VCEI exceptions         : not available

ryzhovau, 2024-03-04:
Could you try to install Entware from scratch using `https://test.entware.net/mipselsf-k3.4`?

ximiximi1, 2024-03-05:
> Could you try to install Entware from scratch using `https://test.entware.net/mipselsf-k3.4`?

Sorry, I don't know what means scratch.
I have installed Entware use [generic.sh](https://bin.entware.net/mipselsf-k3.4/installer/generic.sh).

I saw that https://test.entware.net/mipselsf-k3.4/installer/generic.sh is the same as https://bin.entware.net/mipselsf-k3.4/installer/generic.sh.

ryzhovau, 2024-03-05:
Not exactly the same. `https://test.entware.net/mipselsf-k3.4` contains binaries [without MIPS16 instructions](https://github.com/Entware/Entware/issues/1021), as it was before Feb 2024.

ximiximi1, 2024-03-05:
> Not exactly the same. `https://test.entware.net/mipselsf-k3.4` contains binaries [without MIPS16 instructions](https://github.com/Entware/Entware/issues/1021), as it was before Feb 2024.

I have tried. It does not help.
Still "Illegal instruction".
I checked that my SOC has MIPS16, so maybe this is not a problem about MIPS16.

ximiximi1, 2024-03-05:
I guess maybe something related to kernel syscall.
Does all syscall used in Entware compatible with kernel 5.15?

ryzhovau, 2024-03-07:
Can't reproduce that on ancient Padavan firmware using [this](https://nodejs.org/en/learn/getting-started/introduction-to-nodejs) "hello world" code.

```
# uname -a
Linux WT3020H 3.4.113 #1 Wed Oct 16 13:35:50 MSK 2019 mips GNU/Linux
# cat /proc/cpuinfo 
system type		: MediaTek MT7620 SoC
processor		: 0
cpu model		: MIPS 24KEc V5.0
…
```

ximiximi1, 2024-03-08:
Finally, I found what happened.
OpenWRT disabled the MIPS FPU emulator in kernel as a default setting. That causes this issue.

I have recompiled the OpenWRT image with MIPS FPU emulator in kernel setting and everything goes well now.

Thanks.

closed 2024-03-08 by ryzhovau