Entware/Entware #1027: Tailscaled returns segmentation fault

tERB00, 2024-03-05:
**Tailscaled returns segmentation fault**
[package name]: Tailscale 

**Environment:**
 - Entware feed: [armv7sf-k3.2]
 - Device [Huawei B315 CPE router]
 - Firmware version [v.21.313.05.00.00]

> root@android:/opt/bin # /opt/bin/opkg install tailscale_nohf
> Installing tailscale_nohf (1.58.2-1) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/tailscale_nohf_1.58.2-1_armv7-3.2.ipk
> Installing libgcc (8.4.0-11) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/libgcc_8.4.0-11_armv7-3.2.ipk
> Installing libc (2.27-11) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/libc_2.27-11_armv7-3.2.ipk
> Installing libssp (8.4.0-11) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/libssp_8.4.0-11_armv7-3.2.ipk
> Installing libpthread (2.27-11) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/libpthread_2.27-11_armv7-3.2.ipk
> Installing librt (2.27-11) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/librt_2.27-11_armv7-3.2.ipk
> Installing ca-bundle (20230311-1) to root...
> Downloading http://bin.entware.net/armv7sf-k3.2/ca-bundle_20230311-1_all.ipk
> Configuring libgcc.
> Configuring libc.
> Configuring libpthread.
> Configuring librt.
> Configuring libssp.
> Configuring ca-bundle.
> Configuring tailscale_nohf.
> root@android:/opt/bin # /opt/bin/tailscale up
> failed to connect to local tailscaled; it doesn't appear to be running
> root@android:/opt/bin # /opt/bin/tailscaled
> Segmentation fault
> root@android:/opt/bin #

I tried both nohf and a regular packages - both act the same.

zyxmon, 2024-03-05:
Cannot confirm it. What is your kernel verion?

tERB00, 2024-03-05:
> Cannot confirm it. What is your kernel verion?

`Linux version 3.4.5 (jenkins@localhost) (gcc version 4.6.x-google 20120106 (prerelease) (GCC) ) #1 PREEMPT Tue Aug 21 14:51:51 CST 2018`

zyxmon, 2024-03-05:
```
root@QNAP-D1:~$ opkg list-installed | grep tails
tailscale - 1.58.2-1
root@QNAP-D1:~$ uname -a
Linux QNAP-D1 3.10.20-al-2.5.3 #972 SMP PREEMPT Thu Nov 24 06:55:38 CST 2022 armv7l GNU/Linux
root@QNAP-D1:~$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 74.24
Features        : swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5
.....root@QNAP-D1:~$ tailscale up
failed to connect to local tailscaled; it doesn't appear to be running
root@QNAP-D1:~$ tailscaled
logtail started
Program starting: v1.58.2, Go 1.21.7: []string{"tailscaled"}
LogID: bee1441ee8669b923afd083719c8f45fe162cc46d7b1b7e5e09e05d932d0d459
logpolicy: using UserCacheDir, "/opt/root/.cache/Tailscale"
logpolicy.ConfigFromFile /opt/root/.cache/Tailscale/tailscaled.log.conf: open /opt/root/.cache/Tailscale/tailscaled.log.conf: no such file or directory
logpolicy.Config.Validate for /opt/root/.cache/Tailscale/tailscaled.log.conf: config is nil
--statedir (or at least --state) is required
```

tERB00, 2024-03-05:
> You can try older tailscale versions from archive subfolder http://bin.entware.net/armv7sf-k3.2/archive/

Uggghh... thanks for the advice. I tried a bunch of them and the tailscale_1.36.2-1_armv7-3.2.ipk seems to work. They broke something in the later versions, I guess....

closed 2024-03-05 by tERB00