Entware/Entware #1029: Busybox upgrade failure

sryze, 2024-03-08:
**Describe the bug**
[busybox]: opkg upgrade left busybox in inconsistent state where some commands such as `ls` are missing

```
Upgrading busybox on root from 1.35.0-3 to 1.36.1-1c...
Downloading http://myshare/opkg/packages/busybox_1.36.1-1c_mips-3.4.ipk
jsonfilter was autoinstalled and is now orphaned, removing.
Removing package jsonfilter from root...
Removing obsolete file /opt/lib/upgrade/keep.d/busybox.
free(): invalid pointer
Aborted
```

When I run `opkg upgrade` again I keep getting this error:
```
Collected errors:
 * pkg_get_installed_files: Failed to open //opt/lib/opkg/info/jsonfilter.list: No such file or directory.
```

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4] mips
 - Device [e.g. Asus RT-AX88U router]
 - Firmware version [e.g. asuswrt-merlin]

The-BB, 2024-03-08:
```
``~ # opkg upgrade
Upgrading busybox on root from 1.35.0-3 to 1.36.1-1c...
Downloading http://bin.entware.net/mipssf-k3.4/busybox_1.36.1-1c_mips-3.4.ipk
jsonfilter was autoinstalled and is now orphaned, removing.
Removing package jsonfilter from root...
Removing obsolete file /opt/lib/upgrade/keep.d/busybox.
Configuring busybox.
Warning: '/opt/bin/xargs' is already exists, skipping.
Warning: '/opt/bin/find' is already exists, skipping.
Warning: '/opt/bin/grep' is already exists, skipping.
Warning: '/opt/bin/egrep' is already exists, skipping.
Warning: '/opt/bin/fgrep' is already exists, skipping.
~ # 
~ # ls -1
libs
~ # 
```
need more info

sryze, 2024-03-19:
[opkg-busybox-install-log.txt](https://github.com/Entware/Entware/files/14654005/opkg-busybox-install-log.txt)

closed 2024-03-19 by sryze