Entware/Entware #1033: borgbackup - msgpack error

dajiahao, 2024-03-20:
### Discussed in https://github.com/Entware/Entware/discussions/1032

<div type='discussions-op-text'>

<sup>Originally posted by **dajiahao** March 16, 2024</sup>
I have borgbackup and msgpack installed

I think the related packages are [borgbackup_1.2.4-4_armv7-3.2.ipk] and [python3-msgpack_1.0.7-1_armv7-3.2.ipk]

> borg -V    
borg 1.2.4

but backup to this machine doesn't work, I always get an error about msgpack

> borg serve --info
$LOG ERROR borg.archiver Remote: You do not have a supported version of the msgpack python package i
nstalled. Terminating.
$LOG ERROR borg.archiver Remote: This should never happen as specific, supported versions are requir
ed by our setup.py.
$LOG ERROR borg.archiver Remote: Do not contact borgbackup support about this.

any idea how to fix this? I checked for doubled msgpacks and also reinstalled all python related parts</div>

NRGDEAD, 2024-07-21:
Duplicated [here](https://github.com/Entware/Entware/issues/1036#issue-2230430310), I believe.

ryzhovau, 2025-01-16:
Please, check out new version.

closed 2025-01-16 by ryzhovau