Entware/Entware #1034: Update the manual to install entware on AsusWRT

wootje, 2024-03-29:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

This is a fool proof manual to install Entware on AsusWRT. 

Can you add it to [https://github.com/Entware/Entware/wiki/Install-on-Asus-stock-firmware](url)?

Hopefully it will enable more people to use the Entware repository.

```
1.  format a usb drive as ext3 or etx4 
     use i.e. https://www.easeus.com/partition-manager/epm-free.html

2.  insert the usb drive into the Asus router

3.  install the Download Manager via the web-interface of your Asus router

4.  enable SSH access via the web-interface of your Asus router

5.  setup a SSH connectioon & login using i.e. Putty
    https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html

6.  ipkg update

7.  ipkg install nano

8.  cd /tmp

9  mkdir opt

10. blkid

11. mount /dev/sda /tmp/opt
     run "df" to check if "sda" is your USB drive, if not, change it

12. cd /root

13. touch ~/.wget-hsts

14. chmod 644 .wget-hsts

15. cd /jffs/scripts

16. mkdir scripts

17. chmod 0775 ./scripts

18. vi ./mountOp

19. paste this script:
    #!/bin/sh
    # recreate the opt dir after every reboot
    mkdir /tmp/opt
    # where /dev/sdX# is your usb opt partition
    mount /dev/sdX# /tmp/opt

20. press "esc"

21. type ":x"

22. press ënter"

23 chmod 0775 ./mountOp

24. nvram set script_usbmount="/jffs/scripts/mountOpt"

25. nvram commit

26. paste the srcipt below:

cat << EOF > /tmp/script_usbmount.tmp
if [ \$1 = "/tmp/mnt/MYPART" ]
then
ln -sf \$1 /tmp/opt
/opt/etc/init.d/rc.unslung start
fi
EOF
nvram set script_usbmount="`cat /tmp/script_usbmount.tmp`"

cat << EOF > /tmp/script_usbumount.tmp
if [ \$1 = "/tmp/mnt/MYPART" ]
then
/opt/etc/init.d/rc.unslung stop
fi
EOF
nvram set script_usbumount="`cat /tmp/script_usbumount.tmp`"

nvram commit 
reboot

27. wait until the reboot is finished

28. reconnnect to the router using SSH

29. aarch64 4.1|+:
     wget -O - http://bin.entware.net/aarch64-k3.10/installer/generic.sh | sh
      or
     wget -O - http://bin.entware.net/armv7sf-k3.2/installer/generic.sh | sh

    RT-AC66U B1, RT-AC68U, RT-AC56U, RT-AC87U, RT-AC3200, RT-AC88U, RT-AC3100, RT-AC5300:
     wget -O - http://bin.entware.net/armv7sf-k2.6/installer/generic.sh | sh

    RT-AC58U:
     wget -O - http://bin.entware.net/armv7sf-k3.2/installer/generic.sh | sh

    RT-N66U, RT-AC66U, RT-N16:
     wget -O - http://pkg.entware.net/binaries/mipsel/installer/installer.sh | sh

30. opkg update

31. opkg install nano

32. finished
```

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2024-03-30:
Feel free to edit wiki pages, it's open for everyone.

wootje, 2024-04-15:
Added:
[https://github.com/Entware/Entware/wiki/AsusWRT-install-Entware](url)

closed 2024-08-26 by ryzhovau

wootje, 2024-08-26:
> Added: [https://github.com/Entware/Entware/wiki/AsusWRT-install-Entware](url)
> 
> I can't get the lay-out perfect. I hope it is sufficient enough for people to understand.

Your link is not working.

ryzhovau, 2024-08-27:
As I see, you've successfully edited the wiki page.
Thanks!