Entware/Entware #1042: Add btop package

de-vnull, 2024-05-02:
btop is a more "UI" like view of information than is provided by top or htop. It provides a "dashboard" like view, in a way similar to the GUI applications netdata or other system information applications, but allows for manipulation of active applications.

It's the "dashboard" view that is most compelling ('killer feature'), because it integrates a number of different attributes: processor (name, activity, speed, etc), processes (sortable by various categories), memory (total, cache, swap, etc), disks (usage, swap, etc), network (adapter, upload/download, etc) in a semi-graphical yet keyboard driven view.

I use btop on my other Linux devices (including headless) when I need to monitor the overall picture of the device and think it would be a great addition to Entware, particularly for use on routers (AsusWRT Merlin and FreshTomato).

See a screenshot here: https://www.snbforums.com/threads/addon-development-request-dashboard-views.86951/post-905440

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [AsusWRT Merlin: aarch64 and ARM; FreshTomato ARM; potentially to get multiple users on snbforums involved as well] and get a feedback about it.

ziiiiiiiii, 2025-01-13:
+1
Please add btop package into Entware.

zamarax, 2025-05-09:
I would also really appreciate this :-)

The-BB, 2025-06-10:
This package needs a toolchain with C20++ support.

closed 2025-06-10 by The-BB