Entware/Entware #1046: [Package Request] ajeetdsouza/zoxide (a smarter cd command)

danlsn, 2024-05-08:
**Describe the solution you'd like**

Hi Team,

I'd like to request the addition of zoxide into Entware as it would be super helpful for navigating through my Synology NAS.

It's my understanding that rust packages are now in scope (such as #930) so hopefully this one is suitable. 

- GitHub Repository: [ajeetdsouza/zoxide](https://github.com/ajeetdsouza/zoxide)
- Last release: Feb 21, 2024

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

danlsn, 2024-05-08:
Not sure if this makes it easier, or if adding it to Entware is even necessary (I think it would be) but I managed to install it on my x86_64 Synology DS1821+ from the release binary on GitHub. They also have an aarch64 available for linux

See more:
[zoxide release v0.9.4](https://github.com/ajeetdsouza/zoxide/releases/tag/v0.9.4)

The-BB, 2024-05-09:
Well, try it
http://bin.entware.net/aarch64-k3.10/test/zoxide_0.9.4-1_aarch64-3.10.ipk
http://bin.entware.net/x64-k3.2/test/zoxide_0.9.4-1_x64-3.2.ipk
(optional)
http://bin.entware.net/aarch64-k3.10/test/fzf_0.52.0-1_aarch64-3.10.ipk
http://bin.entware.net/x64-k3.2/test/fzf_0.52.0-1_x64-3.2.ipk

danlsn, 2024-05-12:
Works great! Tested the x64 packages and they're working great on my Synology DS1821+ 

Appreciate it

ryzhovau, 2025-01-16:
Package was included to repo. Thanks for tests!

closed 2025-01-16 by ryzhovau