Entware/Entware #1048: Tar fails to extract symlinks correctly

belsunce, 2024-05-15:
**Describe the bug**
Package: tar 1.34-4

Extracting an archive with tar fails to extract symlinks correctly:
`/opt/bin/tar -xvf syslinux-6.03.tar.gz`
syslinux-6.03/libinstaller/advconst.h: Cannot change mode to rwxrwxrwx: Too many levels of symbolic links

Package: tar
Version: 1.34-4
Depends: xz, libc, libssp, librt, libpthread, libacl, libattr, bzip2
Architecture: armv7-2.6

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k2.6
 - Device: Netgear R7000
 - Firmware version: FreshTomato Version 2024.1
 - opkg version d038e5b6d155784575f62a66a8bb7e874173e92e (2022-02-24)

belsunce, 2024-05-15:
I have figure out that tar 1.32 succeeded to extract symlinks correctly.
Downgrading tar 1.32 restored the functionality. I have downloaded the following package for the router I use:
[tar_1.32-3_armv7-2.6.ipk](https://bin.entware.net/armv7sf-k2.6/archive/tar_1.32-3_armv7-2.6.ipk)

ryzhovau, 2025-01-16:
Just checked 1.35 on yslinux-6.03.tar.gz archive. Looks fine.

closed 2025-01-16 by ryzhovau