Entware/Entware #1049: deluge: installation fails with cryptography version dependency errors

OneCDOnly, 2024-05-20:
**Describe the bug**
deluge: installation fails with cryptography version dependency errors:
```
[~] # opkg install deluge
Installing deluge (2.1.1-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/deluge_2.1.1-1_aarch64-3.10.ipk
Configuring deluge.
Setting default deluge ui to console ... 
Traceback (most recent call last):
  File "/opt/bin/deluge", line 8, in <module>
    sys.exit(start_ui())
             ^^^^^^^^^^
  File "/opt/lib/python3.11/site-packages/deluge/ui/ui_entry.py", line 39, in start_ui
    ui_entrypoints[entrypoint.name] = entrypoint.load()
                                      ^^^^^^^^^^^^^^^^^
  File "/opt/lib/python3.11/site-packages/pkg_resources/__init__.py", line 2515, in load
  File "/opt/lib/python3.11/site-packages/pkg_resources/__init__.py", line 2538, in require
  File "/opt/lib/python3.11/site-packages/pkg_resources/__init__.py", line 829, in resolve
  File "/opt/lib/python3.11/site-packages/pkg_resources/__init__.py", line 875, in _resolve_dist
pkg_resources.ContextualVersionConflict: (cryptography 42.0.2 (/share/CACHEDEV1_DATA/.qpkg/Entware/lib/python3.11/site-packages), Requirement.parse('cryptography<42,>=41.0.5'), {'pyopenssl'})
touch: /opt/etc/deluge/.configured: No such file or directory
Collected errors:
 * pkg_run_script: package "deluge" postinst script returned status 1.
 * opkg_configure: deluge.postinst returned 1.
```
**Environment (please complete the following information):**
 - Entware feed: aarch64-3.10
 - Device: QNAP NAS (TS-230)
 - Firmware version: QTS 5.1.6

<details>
<summary>Installed IPKs:</summary>

```
[~] # opkg list-installed
alsa-lib - 1.2.9-1
ar - 2.41-1
binutils - 2.41-1
boost-python3 - 1.84.0-1
boost-system - 1.84.0-1
ca-bundle - 20230311-1
ca-certificates - 20230311-1
check - 0.15.2-10
column - 2.39.3-1
coreutils - 9.3-1
coreutils-nice - 9.3-1
coreutils-sleep - 9.3-1
coreutils-sort - 9.3-1
coreutils-stty - 9.3-1
coreutils-timeout - 9.3-1
deluge - 2.1.1-1
entware-opt - 227000-3
entware-release - 2024.02-1
entware-upgrade - 1.0-1
ffprobe - 5.1.3-5
file - 5.44-1
findutils - 4.9.0-1a
gcc - 8.4.0-5c
giflib - 5.2.1-1
git - 2.39.2-1
git-http - 2.39.2-1
grep - 3.11-1
htop - 3.3.0-1
ionice - 2.39.3-1
iotop - 0.6-1
jq - 1.7.1-1
lame-lib - 3.100-1
less - 643-1
libarchive - 3.6.2-2
libatomic - 8.4.0-11
libbfd - 2.41-1
libbz2 - 1.0.8-1a
libc - 2.27-11
libctf - 2.41-1
libcurl - 8.6.0-1
libdeflate - 1.18-1
libedit - 20230828-3.1-1
libevent2 - 2.1.12-1
libevent2-core - 2.1.12-1
libevent2-pthreads - 2.1.12-1
libexpat - 2.5.0-1
libexslt - 1.1.39-1
libffi - 3.4.2-2
libffmpeg-full - 5.1.3-5
libfreetype - 2.13.2-1
libgcc - 8.4.0-11
libgdbm - 1.21-2
libgmp - 6.3.0-1
libgnutls - 3.8.3-1
libiconv-full - 1.17-1
libidn2 - 2.3.7-1
libintl-full - 0.21.1-2
libjpeg-turbo - 3.0.2-1
libjson-c - 0.17-1
libleptonica - 1.84.1-1
libltdl - 2.4.7-1
liblzma - 5.4.6-1
libmagic - 5.44-1
libmediainfo - 24.01-1
libmilter-sendmail - 8.16.1-1
libminiupnpc - 2.2.3-1
libnatpmp - 20150609-3
libncurses - 6.4-2a
libncursesw - 6.4-2a
libnettle - 3.9.1-1
libopcodes - 2.41-1
libopenssl - 3.0.13-1
libopenssl-conf - 3.0.13-1
libopenssl-legacy - 3.0.13-1
libopus - 1.4-1
libpcre2 - 10.42-1
libpng - 1.6.39-1
libpsl - 0.21.5-1
libpthread - 2.27-11
libpython3 - 3.11.7-1
libreadline - 8.2-1
librt - 2.27-11
libsensors - 3.6.0-1
libsmartcols - 2.39.3-1
libsqlite3 - 3410200-1
libssp - 8.4.0-11
libstdcpp - 8.4.0-11
libsysfs - 2.1.0-4
libtiff - 4.6.0-1
libtorrent-rasterbar - 2.0.9-1
libunistring - 1.1-1
libutp - 2023-02-14-c95738b1-1
libuuid - 2.39.3-1
libv4l - 1.22.1-1
libwebp - 1.3.2-1
libx264 - 2020-10-26-1
libxml2 - 2.12.4-1
libxslt - 1.1.39-1
libyaml - 0.2.5-1
libzen - 0.4.41-1
libzstd - 1.5.5-1
lm-sensors - 3.6.0-1
locales - 2.27-9
mediainfo - 24.01-1
nano - 7.2-3
objdump - 2.41-1
opkg - 2022-02-24-d038e5b6-2
p7zip - 16.02-3
perl - 5.38.2-1
perlbase-base - 5.38.2-1
perlbase-bytes - 5.38.2-1
perlbase-class - 5.38.2-1
perlbase-config - 5.38.2-1
perlbase-cwd - 5.38.2-1
perlbase-data - 5.38.2-1
perlbase-errno - 5.38.2-1
perlbase-essential - 5.38.2-1
perlbase-fcntl - 5.38.2-1
perlbase-file - 5.38.2-1
perlbase-filehandle - 5.38.2-1
perlbase-getopt - 5.38.2-1
perlbase-i18n - 5.38.2-1
perlbase-integer - 5.38.2-1
perlbase-io - 5.38.2-1
perlbase-list - 5.38.2-1
perlbase-locale - 5.38.2-1
perlbase-params - 5.38.2-1
perlbase-posix - 5.38.2-1
perlbase-re - 5.38.2-1
perlbase-scalar - 5.38.2-1
perlbase-selectsaver - 5.38.2-1
perlbase-socket - 5.38.2-1
perlbase-symbol - 5.38.2-1
perlbase-tie - 5.38.2-1
perlbase-unicore - 5.38.2-1
perlbase-utf8 - 5.38.2-1
perlbase-xsloader - 5.38.2-1
python3 - 3.11.7-1
python3-appdirs - 1.4.4-1
python3-asyncio - 3.11.7-1
python3-attrs - 23.1.0-1
python3-automat - 22.10.0-1
python3-base - 3.11.7-1
python3-bcrypt - 4.0.1-3
python3-certifi - 2024.2.2-1
python3-cffi - 1.16.0-1
python3-cgi - 3.11.7-1
python3-cgitb - 3.11.7-1
python3-chardet - 5.2.0-1
python3-codecs - 3.11.7-1
python3-constantly - 23.10.4-1
python3-cryptodome - 3.18.0-1
python3-cryptography - 42.0.2-1
python3-ctypes - 3.11.7-1
python3-curl - 7.45.2-3
python3-dateutil - 2.8.2-2
python3-dbm - 3.11.7-1
python3-decimal - 3.11.7-1
python3-defusedxml - 0.7.1-1
python3-dev - 3.11.7-1
python3-distutils - 3.11.7-1
python3-email - 3.11.7-1
python3-hyperlink - 21.0.0-1
python3-idna - 3.6-1
python3-ifaddr - 0.2.0-1
python3-incremental - 22.10.0-1
python3-lib2to3 - 3.11.7-1
python3-libtorrent - 2.0.9-1
python3-light - 3.11.7-1
python3-logging - 3.11.7-1
python3-lxml - 4.9.3-1a
python3-lzma - 3.11.7-1
python3-mako - 1.3.0-1
python3-markupsafe - 2.1.3-1a
python3-multiprocessing - 3.11.7-1
python3-ncurses - 3.11.7-1
python3-openssl - 3.11.7-1
python3-orjson - 3.9.13-1
python3-packaging - 23.2-1
python3-pillow - 10.1.0-1
python3-pip - 23.3.1-1
python3-pkg-resources - 69.0.2-1
python3-ply - 3.11-2
python3-psutil - 5.9.5-1
python3-pyasn1 - 0.5.1-1
python3-pyasn1-modules - 0.3.0-1
python3-pycparser - 2.21-2
python3-pydoc - 3.11.7-1
python3-pyopenssl - 23.3.0-1
python3-pytz - 2023.3-1
python3-pyxdg - 0.28-1
python3-readline - 3.11.7-1
python3-regex - 2023.12.25-1
python3-rencode - 1.0.6-3
python3-requests - 2.31.0-1
python3-rpds-py - 0.10.6-1
python3-service-identity - 23.1.0-1
python3-setproctitle - 1.3.3-1
python3-setuptools - 69.0.2-1
python3-six - 1.16.0-1
python3-sqlite3 - 3.11.7-1
python3-twisted - 23.10.0-1
python3-typing-extensions - 4.9.0-1
python3-unittest - 3.11.7-1
python3-urllib - 3.11.7-1
python3-urllib3 - 2.0.4-1
python3-uuid - 3.11.7-1
python3-xml - 3.11.7-1
python3-yaml - 6.0.1-1a
python3-zope-interface - 6.1-1
qt5-core - 5.15.12-1
qt5-network - 5.15.12-1
qt5-sql - 5.15.12-1
qt5-xml - 5.15.12-1
sed - 4.9-1
sqlite3-cli - 3410200-1
sysfsutils - 2.1.0-4
terminfo - 6.4-2a
tesseract - 5.3.3-1a
zlib - 1.3.1-1
zoneinfo-asia - 2024a-1
zoneinfo-core - 2024a-1
zoneinfo-europe - 2024a-1
```

</details>

zyxmon, 2024-05-21:
Confirmed! In case /opt/etc/deluge folder with files was created by the previous version - there is no error.

zyxmon, 2024-05-21:
Fixed via  https://github.com/Entware/entware-packages/commit/1a488ec075c7d194c9a14e500c3c4200435de79a
Please check if it works.

OneCDOnly, 2024-05-21:
Fix confirmed: 
```
[~] # opkg install deluge
Installing deluge (2.1.1-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/deluge_2.1.1-1_aarch64-3.10.ipk
Configuring deluge.
```
Thanks @zyxmon 🤓

closed 2024-05-21 by OneCDOnly