Entware/Entware #1058: Repository is not available

psunenko, 2024-06-14:
Your repository is hosted in russia and it is blocked in Ukraine (soon will be blocked in the rest of the World).

Here is the log:

[I] Jun 14 01:48:13 installer: Info: "ping google.com" ..... OK
[I] Jun 14 01:48:14 installer: Info: "ping bin.entware.net"...
[I] Jun 14 01:48:14 kernel: ACS result: Primary Channel 36, Min Channel Busy = 0, BW = 80
[I] Jun 14 01:48:26 installer: Critical error: Ресурс недоступен. Повторите попытку позже.
[I] Jun 14 01:48:26 installer: Выход из установки.

sending ICMP ECHO request to bin.entware.net...
PING bin.entware.net (158.160.45.54) 56 (84) bytes of data.
--- bin.entware.net ping statistics ---
5 packets transmitted, 0 packets received, 100% packet loss,
0 duplicate(s), time 5004.08 ms.

$ ping bin.entware.net
PING yacloud.entware.net (158.160.45.54) 56(84) bytes of data.
^C
--- yacloud.entware.net ping statistics ---
47 packets transmitted, 0 received, 100% packet loss, time 47108ms

Provider name:	Yandex.Cloud LLC
Provider website:	https://cloud.yandex.ru/
Provider AS:	[200350](https://2ip.me/en/services/information-service/provider-ip?ip=AS200350&a=act)
Location:	[Russian Federation, Moscow](https://2ip.me/en/services/information-service/site-location?ip=158.160.45.54&a=act)
Checking host:	158.160.45.54 (158.160.45.54)

ryzhovau, 2024-06-15:
Can't confirm that. No any downtime on monitoring probes.

closed 2024-06-15 by ryzhovau

alexneo2003, 2024-11-24:
> Can't confirm that. No any downtime on monitoring probes.

Is that your post?))
https://www.snbforums.com/threads/cant-update-install-entware-about-1-mounth.84812/post-838699

psunenko, 2024-11-24:
Hi! This isn't my post, but I experienced the same problem. Solved it by
downloading to my VPS in US and changed the source in the config file.

ryzhovau, 2024-11-25:
There's some stable (but unofficial) [mirrors](https://entware.net/mirrors.html), feel free to use it.

alexneo2003, 2024-11-25:
> There's some stable (but unofficial) [mirrors](https://entware.net/mirrors.html), feel free to use it.

Yep, I'm try to use it but, unfortunately, I still get "critical error"
```bash
[E] Nov 25 20:20:26 ndm: Opkg::Manager: invalid initrc "/opt/etc/initrc": no such file or directory, trying /opt/etc/init.d/. 
[I] Nov 25 20:20:26 installer: [1/5] Начало установки системы пакетов "Entware"...
[I] Nov 25 20:20:26 installer: Info: Раздел пригоден для установки.
[I] Nov 25 20:20:26 installer: Info: Создание каталогов...
[I] Nov 25 20:20:26 installer: [2/5] Загрузка и установка основных пакетов...
[I] Nov 25 20:20:27 installer: Critical error: Во время обновления списка пакетов произошла ошибка.
[I] Nov 25 20:20:27 installer: Выход из установки.
```

alexneo2003, 2024-11-26:
I did it!
Only one mirror `http://mirrors.nju.edu.cn/entware` that not redirected to https working for me