Entware/Entware #1059: Compile mod_sftp for proftpd

Magissia, 2024-06-21:
**Describe the solution you'd like**
Hello,
I would like the module mod_sftp to be compiled with proftpd in the main package, or as an additional package to install with proftpd.

Currently the module is not delivered
```
ash-4.3# /opt/sbin/proftpd -l
Compiled-in modules:
  mod_core.c
  mod_xfer.c
  mod_rlimit.c
  mod_auth_unix.c
  mod_auth_file.c
  mod_auth.c
  mod_ls.c
  mod_log.c
  mod_site.c
  mod_delay.c
  mod_facts.c
  mod_digest.c
  mod_lang.c
```

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [x86-64] and get a feedback about it.

Zulgrib, 2024-08-27:
Also interested and volunteer to test on amd64, mips and armv7l processors.

The-BB, 2025-01-17:
Done! Please, provide feedback.

closed 2025-01-17 by The-BB