Entware/Entware #1060: Request compile a famous Nginx module - ngx_http_subs_filter_module.so

return2alfred, 2024-06-24:
**Dear Entware team, glad if we can compile Nginx module - ngx_http_subs_filter_module.so**

Module Project url is: https://github.com/yaoweibin/ngx_http_substitutions_filter_module
The extreme key feature is: subs_filter derective in nginx config to replace and modify js context
Platform is: Synology DSM 6.2.4, amd64
Repo I've used: https://bin.entware.net/x64-k3.2/

The nginx info:
nginx version: nginx/1.25.3 (x86_64-pc-linux-gnu)
built with OpenSSL 3.0.13 30 Jan 2024
TLS SNI support enabled
configure arguments: --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/opt --exec-prefix=/opt --bindir=/opt/bin --sbindir=/opt/sbin --libexecdir=/opt/lib --sysconfdir=/opt/etc --datadir=/opt/share --localstatedir=/opt/var --mandir=/opt/share/man --infodir=/opt/share/info --crossbuild=Linux::x86_64 --prefix=/opt --conf-path=/opt/etc/nginx/nginx.conf --modules-path=/opt/lib/nginx/modules --error-log-path=/opt/var/log/nginx/error.log --pid-path=/opt/var/run/nginx.pid --lock-path=/opt/var/lock/nginx.lock --http-log-path=/opt/var/log/nginx/access.log --http-client-body-temp-path=/opt/var/lib/nginx/body --http-proxy-temp-path=/opt/var/lib/nginx/proxy --http-fastcgi-temp-path=/opt/var/lib/nginx/fastcgi --http-scgi-temp-path=/opt/var/lib/nginx/scgi --http-uwsgi-temp-path=/opt/var/lib/nginx/uwsgi --with-cc=x86_64-openwrt-linux-gnu-gcc --with-cc-opt='-I/home/me/E/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/usr/include -I/home/me/E/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/include -I/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/include -I/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/include -O2 -pipe -fomit-frame-pointer -fno-caller-saves -fhonour-copts -ffunction-sections -fdata-sections -I/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/include -I/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/include -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-Wl,--dynamic-linker=/opt/lib/ld-linux-x86-64.so.2 -Wl,-rpath=/opt/lib -L/home/me/E/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/usr/lib -L/home/me/E/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/lib -fuse-ld=bfd -Wl,--gc-sections -L/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/lib -Wl,-rpath-link=/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/lib -L/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/lib -Wl,-rpath-link=/home/me/E/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/lib' --without-http_upstream_zone_module --with-compat --with-http_ssl_module --with-ipv6 --with-http_gzip_static_module --with-http_stub_status_module --with-http_flv_module --with-http_dav_module --with-http_auth_request_module --with-http_v3_module --with-http_v2_module --with-http_realip_module --with-http_secure_link_module --with-http_sub_module --with-stream=dynamic --with-stream_ssl_module --with-stream_ssl_preread_module --with-stream_realip_module --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-naxsi/naxsi_src --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-geoip2 --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-lua --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-rtmp --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-dav-ext --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-brotli --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-headers-more --add-dynamic-module=/home/me/E/build_dir/target-x86_64_glibc-2.27/nginx-ssl/nginx-1.25.3/nginx-mod-ts

- [X] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

return2alfred, 2024-06-25:
Hi Dears, can I have any replies? Positive or negative, thanks a lot!

ryzhovau, 2026-11-04:
Nginx package came form upstream with no any changes, to easier further sync.
No plans to change it, sorry.

closed 2026-11-04 by ryzhovau