Entware/Entware #1064: CVE-2024-6387: *critical* regreSSHion vulnerability likely in latest available entware openSSH packages (sshd)

deekayw0n, 2024-07-04:
Description: On July 1, 2024, a critical signal handler race condition vulnerability was disclosed in OpenSSH servers (sshd) on glibc-based Linux systems. This vulnerability, called RegreSSHion and tracked as [CVE-2024-6387](https://ubuntu.com/security/CVE-2024-6387), can result in unauthenticated remote code execution (RCE) with root privileges. This vulnerability has been rated High severity ([CVSS 8.1](https://www.cve.org/CVERecord?id=CVE-2024-6387)).

This vulnerability impacts the following OpenSSH server versions:

Open SSH version between 8.5p1-9.8p1
Open SSH versions earlier than 4.4p1, if they’ve not backport-patched against [CVE-2006-5051](https://nvd.nist.gov/vuln/detail/CVE-2006-5051) or patched against [CVE-2008-4109](https://nvd.nist.gov/vuln/detail/CVE-2008-4109)

**Latest Entware packages likely impacted (but potentially others):**
- openssh-server - 9.6p1-1 - OpenSSH server.
- openssh-server-pam - 9.6p1-1 - OpenSSH server (with PAM support).
 
**Environment (please complete the following information):**
- glibc-based Linux systems with impacted version of sshd

**More general information:**
- https://www.zdnet.com/article/over-14m-servers-may-be-vulnerable-to-opensshs-regresshion-rce-flaw-heres-what-you-need-to-do/

SaeedDev94, 2024-07-16:
No update for openssh ?!

The-BB, 2024-07-16:
https://github.com/openwrt/packages/commit/75674f0439ee497bc6b77222a23e3974d150be89 => https://github.com/Entware/entware-packages/commit/90178eb25926917a751f2b01fe346986c78e826d

SaeedDev94, 2024-07-16:
> [openwrt/packages@75674f0](https://github.com/openwrt/packages/commit/75674f0439ee497bc6b77222a23e3974d150be89) => [Entware/entware-packages@90178eb](https://github.com/Entware/entware-packages/commit/90178eb25926917a751f2b01fe346986c78e826d)

Thanks but when can we get the new update ?

deekayw0n, 2024-07-16:
Hi folks, thanks kindly @The-BB for kicking the tires on this one! hex 9.8p1 pkg hash looks good based on official rls notes: https://www.openssh.com/releasenotes.html. 

For clarity and the casual observers that might still think this is an impacted version, the range I provided above was not boundary inclusive at least on the upper bound. This is the fixed version.

Looking fwd to it getting pushed out.  Cheers!

SaeedDev94, 2024-08-04:
I kindly ask for faster updates for such these critical security issues like this, thanks

maurerr, 2024-08-04:
> I kindly ask for faster updates for critical security issues like this, Thanks

you can always build from source - or pay someone to do it - or patiently wait for the wonderful people from entware to produce a release

ryzhovau, 2024-08-09:
Binaries will arrive soon.

closed 2024-08-09 by ryzhovau