Entware/Entware #1066: Entware/docker error

maurerr, 2024-07-12:
@ryzhovau I'm hitting the following error with docker entware:
```
me@4c0f6ccf0de9:/data/entware$ make package/tmux/compile V=s
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
xargs: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by xargs)
xargs: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by xargs)
xargs: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by xargs)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
xargs: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by xargs)
make[2]: Entering directory '/data/entware/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/data/entware/scripts/config'
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by find)
find: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by find)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by sed)
sed: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sed)
time: target/linux/prereq#0.07#0.03#0.10
make[1]: Entering directory '/data/entware'
make[1]: *** No rule to make target 'package/tmux/compile'.  Stop.
make[1]: Leaving directory '/data/entware'
make: *** [/data/entware/include/toplevel.mk:248: package/tmux/compile] Error 2
```
does it needs debian 12?

closed 2024-07-19 by maurerr

ryzhovau, 2024-07-19:
Sorry for so-long-keeping-you-waiting, we are in the middle of another update.
Please, take a look at our Docker image here https://github.com/Entware/docker/blob/main/Dockerfile

maurerr, 2024-07-19:
no worries - the issue was at my end