Entware/Entware #1070: zerotier_1.14.0-1_aarch64-3.10.ipk NOT working with CoreELEC 21.0-Omega (Amlogic-ng.arm; S905X3)

roisiano, 2024-08-09:
**Describe the bug**
[package name]: 
zerotier_1.14.0-1_aarch64-3.10.ipk
from:
https://bin.entware.net/aarch64-k3.10/

I was using zerotier_1.12.2-2_aarch64-3.10.ipk (https://bin.entware.net/aarch64-k3.10/archive/zerotier_1.12.2-2_aarch64-3.10.ipk) and, after upgrading all entware packages:
opkg update
opkg upgrade

version zerotier_1.14.0-1_aarch64-3.10.ipk was installed, but NOT working.

 zerotier-cli status
Error connecting to the ZeroTier service: connection failed

Please check that the service is running and that TCP port 9993 can be contacted via 127.0.0.1.

I'm using this with CoreELEC 21.0-Omega (Amlogic-ng.arm), in a S905X3 TV-Box (X96 Max+).

After this, I "downgraded" it:
opkg remove zerotier

I downloaded previous version "manually":
https://bin.entware.net/aarch64-k3.10/archive/zerotier_1.12.2-2_aarch64-3.10.ipk

I put it in:
/storage/downloads

and installed it manually:
cd /storage/downloads
opkg install zerotier_1.12.2-2_aarch64-3.10.ipk

and, afther that, it gives this error:
zerotier-cli status
zerotier-cli: error while loading shared libraries: libminiupnpc.so.17: cannot open shared object file: No such file or directory

So, what I did it was restore a previous backup that I had with zerotier_1.12.2-2_aarch64-3.10.ipk working right.
It works OK again (but with version zerotier_1.12.2-2).

I tried again all the process (opkg upgrade; opkg update) with same results.
Restored again the backup and working.

So, I think there is a problem with last zerotier version.

Thanks in advance and kind regards

ryzhovau, 2024-08-10:
Sounds familiar:)
https://github.com/Entware/Entware/wiki#%EF%B8%8F-package-upgrade-notes

zyxmon, 2024-08-10:
@roisiano can you give the steps to reproduce problem with the new version of zerotier.

roisiano, 2024-08-10:
> @roisiano can you give the steps to reproduce problem with the new version of zerotier.

Hello:

The "problem" is that when I do:
opkg update
opkg upgrade

NO error is detected:

# opkg update
Downloading http://bin.entware.net/aarch64-k3.10/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware

# opkg upgrade
Upgrading terminfo on root from 6.4-2a to 6.4-3...
Downloading http://bin.entware.net/aarch64-k3.10/terminfo_6.4-3_aarch64-3.10.ipk
Upgrading opkg on root from 2022-02-24-d038e5b6-2 to 2022.02.24~d038e5b6-2...
Downloading http://bin.entware.net/aarch64-k3.10/opkg_2022.02.24~d038e5b6-2_aarch64-3.10.ipk
Upgrading libiperf3 on root from 3.16-1 to 3.17.1-2...
Downloading http://bin.entware.net/aarch64-k3.10/libiperf3_3.17.1-2_aarch64-3.10.ipk
Upgrading bash on root from 5.2.15-1 to 5.2.21-1...
Downloading http://bin.entware.net/aarch64-k3.10/bash_5.2.21-1_aarch64-3.10.ipk
**Upgrading zerotier on root from 1.12.2-2 to 1.14.0-1...
Downloading http://bin.entware.net/aarch64-k3.10/zerotier_1.14.0-1_aarch64-3.10.ipk**
Upgrading tailscale on root from 1.58.2-1 to 1.68.2-1...
Downloading http://bin.entware.net/aarch64-k3.10/tailscale_1.68.2-1_aarch64-3.10.ipk
Upgrading libnatpmp on root from 20150609-3 to 20230423-1...
Downloading http://bin.entware.net/aarch64-k3.10/libnatpmp_20230423-1_aarch64-3.10.ipk
Removing obsolete file /opt/lib/libnatpmp.so.20150609.
Upgrading libpcsclite on root from 2.0.1-1 to 2.2.3-2...
Downloading http://bin.entware.net/aarch64-k3.10/libpcsclite_2.2.3-2_aarch64-3.10.ipk
Removing obsolete file /opt/lib/libpcsclite.so.1.0.0.
Upgrading entware-release on root from 2024.02-1 to 2024.07-1...
Downloading http://bin.entware.net/aarch64-k3.10/entware-release_2024.07-1_all.ipk
Upgrading libminiupnpc on root from 2.2.3-1 to 2.2.8-1...
**Downloading http://bin.entware.net/aarch64-k3.10/libminiupnpc_2.2.8-1_aarch64-3.10.ipk
Removing obsolete file /opt/lib/libminiupnpc.so.17.
Removing obsolete file /opt/lib/libminiupnpc.so.2.2.3.**
Upgrading libncurses on root from 6.4-2a to 6.4-3...
Downloading http://bin.entware.net/aarch64-k3.10/libncurses_6.4-3_aarch64-3.10.ipk
Upgrading libncursesw on root from 6.4-2a to 6.4-3...
Downloading http://bin.entware.net/aarch64-k3.10/libncursesw_6.4-3_aarch64-3.10.ipk
Upgrading findutils on root from 4.9.0-1a to 4.10.0-1...
Downloading http://bin.entware.net/aarch64-k3.10/findutils_4.10.0-1_aarch64-3.10.ipk
Upgrading htop on root from 3.3.0-1 to 3.3.0-2...
Downloading http://bin.entware.net/aarch64-k3.10/htop_3.3.0-2_aarch64-3.10.ipk
Upgrading ca-bundle on root from 20230311-1 to 20240203-1...
Downloading http://bin.entware.net/aarch64-k3.10/ca-bundle_20240203-1_all.ipk
Upgrading libnl-tiny on root from 2023-12-05-965c4bf4-1 to 2023.12.05~965c4bf4-1...
Downloading http://bin.entware.net/aarch64-k3.10/libnl-tiny_2023.12.05~965c4bf4-1_aarch64-3.10.ipk
Upgrading iperf3 on root from 3.16-1 to 3.17.1-2...
Downloading http://bin.entware.net/aarch64-k3.10/iperf3_3.17.1-2_aarch64-3.10.ipk
Upgrading libopenssl on root from 3.0.13-1 to 3.0.14-2...
Downloading http://bin.entware.net/aarch64-k3.10/libopenssl_3.0.14-2_aarch64-3.10.ipk
Upgrading iftop on root from 2018-10-03-77901c8c-2 to 2018.10.03~77901c8c-2...
Downloading http://bin.entware.net/aarch64-k3.10/iftop_2018.10.03~77901c8c-2_aarch64-3.10.ipk
Configuring terminfo.
Configuring opkg.
Configuring libiperf3.
Configuring libnl-tiny.
Configuring libncursesw.
Configuring libncurses.
Configuring bash.
Configuring libminiupnpc.
Configuring libnatpmp.
**Configuring zerotier.**
Configuring ca-bundle.
Configuring tailscale.
Configuring libpcsclite.
Configuring entware-release.
Configuring findutils.
Configuring htop.
Configuring iperf3.
Configuring libopenssl.
Configuring iftop.

The result of the update is that after updating, zerotier does NOT starts. Neither after rebooting the TV-box.

#  zerotier-cli status
Error connecting to the ZeroTier service: connection failed

Please check that the service is running and that TCP port 9993 can be contacted via 127.0.0.1.

In my opinion, it could be a problem due to the removal of old versions of libminiupnpc, but it is just a theory.

zyxmon, 2024-08-10:
@roisiano what steps I am to perform to check that new version does not work? Configuration?

roisiano, 2024-08-10:
> @roisiano what steps I am to perform to check that new version does not work? Configuration?

Well... I think you have to register in zerotier:
https://my.zerotier.com/

Create a network, etc.

I followed this tutorial (sorry, in Spanish):
https://telegra.ph/ZeroTier-para-CE-v%C3%ADa-entware-09-17

and it worked without issues until this moment, except this previous issue that I reported 2 years ago:
https://github.com/Entware/Entware/issues/785

No "special" configuration at all.
1) Register in Zerotier and create your network
2) Use this script:
https://gist.githubusercontent.com/meoso/b25bd410c8a54a1a013f0cc2d72b12ee/raw/b1d19b909f6e94870f710fda1ecf706b3204cb32/S90zerotier-one.sh
Copy the script as:
/opt/etc/init.d/S90zerotier-one.sh

3) Then:
cd /opt/etc/init.d
chmod 755 S90zerotier-one.sh
./S90zerotier-one.sh

4) Reboot

5) Join your network (in this example, ef56tfgtrfsw):
zerotier-cli join ef56tfgtrfsw

6) It works?

Kind regards

zyxmon, 2024-08-10:
I have tested zerotier 1.14 on my aarch64 NAS and it works:

```
root@QNAP128A:~$ opkg install zerotier
Installing zerotier (1.14.0-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/zerotier_1.14.0-1_aarch64-3.10.ipk
Configuring zerotier.
root@QNAP128A:~$ zerotier-one -d
root@QNAP128A:~$ Starting Control Plane...
Starting V6 Control Plane...
....
<in another cli session>
# zerotier-cli join xxxxxxx
200 join OK

```

roisiano, 2024-08-10:
OK.

I did now:

opkg remove zerotier
Removing package zerotier from root..

opkg remove libminiupnpc
Removing package libminiupnpc from root...

Also removed:
/opt/etc/init.d/S90zerotier-one.sh

reboot

And then:
opkg install zerotier
Installing zerotier (1.14.0-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/zerotier_1.14.0-1_aarch64-3.10.ipk
Installing libminiupnpc (2.2.8-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libminiupnpc_2.2.8-1_aarch64-3.10.ipk
Configuring libminiupnpc.
Configuring zerotier.

zerotier-one -d
CoreELEC3M:~ # Starting Control Plane...
Starting V6 Control Plane...

zerotier-cli join (here my network)
0 join connection failed

reboot

Then I made 
/opt/etc/init.d/S90zerotier-one.sh
stuff.

Reboot.

zerotier-cli join (here my network)
0 join connection failed

zerotier-cli status
Error connecting to the ZeroTier service: connection failed

Please check that the service is running and that TCP port 9993 can be contacted via 127.0.0.1.

I do NOT know exactly why it does NOT work, but with new version, it does NOT work for me. With the previous one, it works.

Kind regards

zyxmon, 2024-08-10:
May be some system calls are missing in CoreElec. Some further investigations should be performed (with strace or something else). I am closing this ticket. More info is needed.

closed 2024-08-10 by zyxmon

m-zheng, 2024-08-20:
@roisiano I ran into the same error. After spending a few hours of debuging and testing, please follow the solution below to fix the problem.

TLDR: downgrade `libminiupnpc` and `zerotier` to the previous version .
1. remove `libminiupnpc` with `opkg remove libminiupnpc --force-depends`
2. remove `zerotier` with `opkg remove zerotier --force-depends`
3. download the previous version for libminiupnpc with `wget "http://bin.entware.net/aarch64-k3.10/archive/libminiupnpc_2.2.3-1_aarch64-3.10.ipk"`. Based on your device's chip, you may need to visit `bin.entware.net` to find the right package.
5. install the old version with `opkg install libminiupnpc_2.2.3-1_aarch64-3.10.ipk`
6. download the previous version for zerotier with `wget "http://bin.entware.net/aarch64-k3.10/archive/zerotier_1.12.2-2_aarch64-3.10.ipk"`. 
5. install the old version with `opkg install zerotier_1.12.2-2_aarch64-3.10.ipk`
7. enjoy

roisiano, 2024-08-20:
Thanks, @m-zheng  

What is your device and operating system?

In my case, it was even simpler.
I restored a previous backup with zerotier_1.12.2-2 and it works again.

By the way, I had this error after upgrade in other CoreELEC devide (Vontar X4, S905X4, CoreELEC 21.1-ne), so it is NOT a problem with a single device.
Returning to zerotier_1.12.2-2, it works again.

Kind regards

m-zheng, 2024-08-20:
Asus router with merlin firmware.

roisiano, 2024-08-21:
> Asus router with merlin firmware.

Then, this dos NOT seem a specific CoreELEC problem.

roisiano, 2024-11-12:
Thanks to the user @hav , there is a "patch" to use latest version of Zerotier in CoreELEC without problems.
You can create a service and enable it.

Something like this (zerotier-one.service):

`
[Unit]
Description=ZeroTier One
After=network-online.target
Wants=network-online.target

[Service]
ExecStart=/storage/.opt/bin/zerotier-one
Restart=always
RestartSec=10
StartLimitInterval=30
StartLimitBurst=20
KillMode=process

[Install]
WantedBy=kodi.target
`