Entware/Entware #1073: Impossible to clone any git repo with "git-http - 2.45.2-1" package

ilyaryabikin, 2024-08-11:
**Describe the bug**
Hi!

[git-http - 2.45.2-1]: Currently, it is not possible to clone any Git repository. After executing the `git clone` command, I receive the following errogit clone https://github.com/Entware/Entware.gitr: `fatal: could not create leading directories of '<repo>/.git'`. The error provides no additional details or reasons. I attempted to specify the directory to clone into explicitly and installed `coreutils-mkdir` instead of using the busybox applet, but the issue persists. The previous version of this package (e.g., git_2.39.2-1) works correctly.

**Environment (please complete the following information):**
 - Entware feed: [mipsel-3.4_kn]
 - Device: [Keenetic Hopper (KN-3810)]
 - Firmware version: [KeeneticOS 4.1.7]

**Steps to reproduce:**
1. `opkg install git-http`
2. `git clone https://github.com/Entware/Entware.git` or `mkdir entware && git clone https://github.com/Entware/Entware.git entware` 
3. `fatal: could not create leading directories of 'Entware/.git'`

ilyaryabikin, 2024-08-11:
Sorry, just noticed the https://github.com/Entware/Entware/issues/1072 marked as closed. Closing this issue

ilyaryabikin, 2024-08-11:
Might be related to https://github.com/Entware/Entware/issues/1072, but different error message

zyxmon, 2024-08-11:
It is already fixed via https://github.com/Entware/entware-packages/commit/ddd8b9323242832f066f3496389cc02f0e6f6021

closed 2024-08-11 by zyxmon

ilyaryabikin, 2024-08-11:
Package wasn't available for me at the moment of creating this issue. Now fixed after upgrading, thanks!