Entware/Entware #1076: gcc package does not contain gfortran compiler

MisesEnForce, 2024-08-13:
The gcc package provides a C and a C++ compilers but no other compiler (as GNU official gcc does), especially not the fortran compiler gfortran. It would be nice to have it.

I'm ready to test added package on my platform [my use case is synology DS920+] and get a feedback about it.

ryzhovau, 2024-08-13:
It's a part of toolchain. We keep it in current state to avoid any of compatibility issues.

closed 2024-08-13 by ryzhovau

MisesEnForce, 2024-08-13:
> It's a part of the toolchain. We keep it in its current state to avoid any compatibility issues. Won't add it, sorry

What do you mean with "it's a part of the toolchain" ?

ryzhovau, 2024-08-14:
Ah, now I get it! You mentioned the native GCC package, not the host GCC compiler for cross-compilation. Sorry. 

We provide native compilation tools [as is](https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation) and are not planning to develop them.

OlorinOLorien, 2024-08-15:
@ryzhovau Hello. I actually have the same "issue" : we have a DS920+ at our company we need to run legacy fortran code on. I understand you won't provide gfortran as openwrt doesn't provide it, fair enough. So I tried to build it from source on the DS920+, using C and C++ compiler still provided by the install of your native gcc package. First I need to build gmp, for which the build went like : 

`
wget https://ftp.gnu.org/gnu/gmp/gmp-6.3.0.tar.xz
tar -Jxf gmp-6.3.0.tar.xz
cd gmp-6.3.0/
mkdir build && cd build
../configure --prefix=/usr/local/lvm/gmp-6.0.0 --enable-cxx
make
`

The make yields the error :

`./libtool: line 2642: func__fatal_error: command not found`

That's a huge typo as it should be func_fatal_error instead of func__fatal_error. After correction and remake I finally have the following error, the real serious error :

`libtool:   error: unsupported hardcode properties See the libtool documentation for more information. Fatal configuration error.`

triggered by (the execution of) the command :

`./libtool  --tag=CXX   --mode=link g++  -O2 -pedantic -fomit-frame-pointer -m64 -mtune=k8 -march=k8   -version-info 11:0:7  -o libgmpxx.la -rpath /usr/local/lvm/gmp-6.3.0/lib cxx/dummy.lo cxx/isfuns.lo cxx/ismpf.lo cxx/ismpq.lo cxx/ismpz.lo cxx/ismpznw.lo cxx/limits.lo cxx/osdoprnti.lo cxx/osfuns.lo cxx/osmpf.lo cxx/osmpq.lo cxx/osmpz.lo libgmp.la`

Would you have any idea about what is happening here ?

Other strategy : I could cross build all GCC compilers (gcc, g++, gfortran etc) from my ubuntu 24.04 on WSL2. Do you have any experience in the parameters (arch, platform etc) for Entware running on a DS920+ equipped with an Intel Celeron J4125 ?